ARMmbed/mbed-os #5351: STM32 CAN filter

pilotak, 2017-10-20:
I would like to set up CAN filtering so that i receive only `ID=0xFF0080` and don't care about last byte (0x80) and bytes before so the mask i expect would be `MASK=0xFFFF00`, but when i run following code and start sending CAN messages I will receive all messages `0xFF00xx`, `0xFF01xx`, `0xFF05xx` etc. however messages below `0xFF00xx` are rejected so the filter partly works.

Also as well the `handle` param in `can.filter() ` doens't work with 0 but works with 14, why is that? is it because it's actualy `CAN2` so `CAN1` has 0-13 and `CAN2` has 14-27?

What am I doing wrong?

```cpp
#include <mbed.h>
CAN can(PB_12, PB_13);
DigitalOut can_sleep(PB_14, 0);

#define ID   0xFF0080
#define MASK 0xFFFF00

int main()
{
    can.frequency(250000);
    can.mode(CAN::Normal);
    can.filter(ID, MASK, CANExtended, 14);

    // how filters works
    if ( (0x18FF0080 & MASK) == (ID & MASK)) // will pass
        printf("0xFF0080 ok\n");
    else
        printf("0xFF0080 failed\n");

    if ( (0x18FF0580 & MASK) == (ID & MASK)) // will fail
        printf("0FF0580 ok\n");
    else
        printf("0xFF0580 failed\n");

    CANMessage msg;

    while (1) {
        if (can.read(msg)) 
            printf("Message received ID: %X\n", msg.id);

        wait(0.2);
    }
}
```

Thank you in advance

### Description
- Type: Bug | Question
- Priority: Minor

**Target**
STM32

**Toolchain:**
GCC_ARM

0xc0170, 2017-10-27:
cc @bcostm @adustm @LMESTM @jeromecoutant

bcostm, 2017-10-30:
I am not a CAN expert but I tried to reproduce your example.

I have programmed a TX CAN device which sends continuous messages with ID equal from 0xFF0080 to 0xFF0280. 

On the RX CAN device:

If I want to receive messages with ID starting with FF and ending with either 80 or 180 or 280, I programmed the ID to 0xFF0080 and the MASK to 0xFF00FF.

If I want only messages with ID starting with 0xFF and ending with 0xB, I programmed on the RX device the ID to 0xFF000B and the mask to 0xFF000F.

If I want only messages with ID strictly equal to 0xFF0080, I programmed on the RX device the ID to 0xFF0080 and the mask to 0xFFFFFF.

Hope this helps.

pilotak, 2017-11-03:
Here is what i have
```cpp
#include <mbed.h>
CAN can(PB_12, PB_13);
DigitalOut can_sleep(PB_14, 0);

#define ID   0xFF0080
#define MASK 0xFFFFFF

int main()
{
    can.frequency(250000);
    can.mode(CAN::Normal);
    can.filter(ID, MASK, CANExtended, 14);

    CANMessage msg;

    while (1) {
        if (can.read(msg)) {
            printf("Message received ID: %X\n", msg.id);
        }
    }
}
```

I'm sending on TX side 0xFF0080, 0xFF0180, 0xFF0580
In theory as you saying i should be receiving only 0xFF0080 but i'm receiving all of above. Could you kindly provide me with your working code? BTW my board is NUCLEO_F412ZG

bcostm, 2017-11-06:
On my side I tried on two NUCLEO_F446RE boards.

Here my test code:
[CAN.zip](https://github.com/ARMmbed/mbed-os/files/1446343/CAN.zip)

One difference I can see is that you use CANExtended when you set the CAN filter and  I use CANStandard. I don't know if this could have an impact ?

bcostm, 2018-01-04:
@pilotak Can you please try to modify the can_filter function in can_api.c file as described in #5765 and tell me if it solves your issue ? Thanks.

pilotak, 2018-01-08:
I applied your patch and it works perfectly I used `CAN1` and `handle`  0

closed 2018-01-18 by pilotak