ARMmbed/mbed-os #5357: Uvision5: export from within a sub folder

chrissnow, 2017-10-21:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
Any
**Toolchain:**
ARM, possibly others

**Expected behavior**
Export's a valid project, or refuses to export.
**Actual behavior**
Export's a project but the paths are all wrong

**Steps to reproduce**
```
mbed import mbed-os-example-blinky
cd mbed-os-example-blinky
mbed export -i uvision5
```
Works as expected and creates a project in the normal place.
sample file ref.
```
<FilePath>.\mbed-os\drivers\BusIn.cpp</FilePath>
```

However move into a subdirectory.
```
cd mbed-os
mbed export -i uvision5
```
Which is successful and creates the project in the normal place but..
Sample file ref
```
<FilePath>..\mbed-os\drivers\BusIn.cpp</FilePath>
```
Notice the ..\ which should be .\

0xc0170, 2017-10-27:
cc @theotherjimmy

theotherjimmy, 2017-10-27:
> Notice the ..\ which should be .\

Agreed. I'll try to reproduce this next week, and see what I can do about it.

theotherjimmy, 2017-11-01:
Reproduced. I'll see what I can do about it now.

theotherjimmy, 2017-12-07:
Sorry for the delay. I have not had time to look into why this is happening with greater detail.

theotherjimmy, 2018-06-20:
I think #7183 Actually fixes this bug. I have been testing with this test case, anyway.

PR #7183 (2018-07-17): Tools: Extract resources object and implement incremental scan
### Description

I found that the resource object being part of the tool chain, and depending on 
tool chain infrastructure was a bit odd, so I extracted resources from tool 
chains. The resulting code is quite a bit more clear in it's purpose and can be a 
tad smaller (extra documentation and formatting excluded). I made the resources 
object fully incremental, meaning that the scan stops early otfen for targets, 
toolchains and features, and can easily pick itself back up where it left off.

A side effect of fully incremental sca [... cut]