ARMmbed/mbed-os #5364: Reduce HAL code duplication

jorisa, 2017-10-23:
With the crazy amount of targets that are now supported also comes a crazy amount of duplicated code. Now that the mbed build system is getting more and more sophisticated, would it be an idea to create a HAL for a specific peripheral, and than bind these to a specific target using the target json?

I'm not sure how well this would work out in practice as there might be variations between different microcontrollers implementations, but I am expecting that the same peripherals get re-used across a family.

As en example I looked at the C_CAN implementation here:
[TARGET_NXP\TARGET_LPC15XX\can_api.c](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NXP/TARGET_LPC15XX/can_api.c)
and
[TARGET_NXP\TARGET_LPC11XX_11CXX\TARGET_LPC11CXX\can_api.c](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NXP/TARGET_LPC11XX_11CXX/TARGET_LPC11CXX/can_api.c)

Only diff [here](http://www.mergely.com/FfBaNLWu/) for your convenience

It seems like they are functionally identical, except that someone has gone through the effort to make one of them more pretty.

Could something like this work?

For example:
```
"LPC1549": {
    "supported_form_factors": ["ARDUINO"],
    "core": "Cortex-M3",
    "default_toolchain": "uARM",
    "extra_labels": ["NXP", "LPC15XX"],
    "supported_toolchains": ["uARM", "GCC_CR", "GCC_ARM", "IAR"],
    "inherits": ["LPCTarget"],
    "detect_code": ["1549"],
    "device_has": {
        "ANALOGIN" : {
            "driver": "/targets/NXP/peripheral/ADC_12BIT_V1",
            "instance": {
                "ADC0": "LPC_ADC0_BASE",
                "ADC1": "LPC_ADC1_BASE",
            }
        },
        "CAN" :  {
            "driver": "/targets/NXP/peripheral/C_CAN",
            "instance": {
                "CAN0": "LPC_C_CAN0_BASE",
            }
        }
    },
    "default_lib": "small",
    "release_versions": ["2"],
    "device_name": "LPC1549JBD64"
},
```

I think the main two advantages would be:
* More efficient to refactor code, less likely that bugs only get fixed for a specific targets 
* Easier to add support to even more microcontroller variants (that are very similar to parts already supported, but slightly different configuration)

0xc0170, 2017-10-23:
Usually this is true for MCU families (even acros some families within the same vendor). This use-case is covered in SDK that come from a vendor (very often). For instance mcuxpresso from NXP, emblib drivers for EFM32. We can look at Silicon labs efm32 HAL, STM32 HAL or NXP MCUXpresso HAL.

As those examples (many LPC targets that were ported some time ago), they are targets that are older and back then we were porting targets by manually implementing entire HAL (not reusing SDK drivers for instance).

cc @bulislaw @c1728p9

jorisa, 2017-10-29:
I just looked at the STM32, and I indeed see there is a lot more reuse as a result from inheritance. However the same issue remains, as I see the an functional identical CAN driver for each family (F0/F1/F2/...) all with some slight tweaks.

It looks like they are different versions of the SDK, did the different families get imported from different HAL versions? Regardless would it be an idea to create a common source folder one level up? (in this example TARGET_STM32/device/stm32xxx_hal_can.c) ?

0xc0170, 2017-10-29:
> It looks like they are different versions of the SDK, did the different families get imported from different HAL versions? Regardless would it be an idea to create a common source folder one level up? (in this example TARGET_STM32/device/stm32xxx_hal_can.c) ?

That is a question for SDK driver owners/implementators. In this case, it is correct that each family has own set of device files (SDK drivers).
@jeromecoutant @LMESTM @adustm might help here

jeromecoutant, 2017-11-02:
@bcostm

adustm, 2017-11-06:
Hello,
Let me answer for STM32 team.
The small differences are usually important to adapt to the differences between 2 hardwares.

You have to know that the code located in targets/TARGET_STM/TARGET_STM32XX/device are the official drivers from STM32Cube emmbedded software. 
As STteam working on mbed, we make the match between mbed HAL and STM drivers, and make the effort of factoring code wherever possible (the common code is located in targets/TARGET_STM/xxx.c files). Our objective is to avoid modifications in the official STM code, to benefit from STM validation work.

Kind regards
Armelle

closed 2018-10-02 by adbridge