ARMmbed/mbed-os #5367: [UBLOX_EVK_ODIN_W2] blinky abnormal

andrewc-arm, 2017-10-24:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
UBLOX_EVK_ODIN_W2

**Toolchain:**
GCC_ARM

**Toolchain version:**
5.4.1

**mbed-cli version:** 
(`mbed --version`)
1.2.0

**mbed-os sha:**
(`git log -n1 --oneline`)
Tried both mbed-os master (1566395) and the one from mbed-os-example-blinky's lib (6e0d01c). They all fail.

**Expected behavior**
Only LED1 (Red) should blink.

**Actual behavior**
It blinks in this pattern.
Green -> Yellow -> Green -> Yellow -> ...

**Steps to reproduce**
Get the latest mbed-os-example-blinky.
Run it for UBLOX_EVK_ODIN_W2.

Looks like we have to turn off LED2 from the beginning.
Why is it turned on from the start?

chrissnow, 2017-10-24:
Looking at the schematic the LED will be on with logic low which is default.
relates to #4035

0xc0170, 2017-10-24:
cc @andreaslarssonublox @andreaspeterssonublox

andreaslarssonublox, 2017-10-25:
Hi,
For the ODIN board we could probably add some default initialization of the LED pins somewhere in the initialization of the HAL. There is a function called HAL_MspInit that is used for this purpose I think. The problem is that it does not seem to be possible to override it((weak in stm32f4xx_hal.c)) because it's defined in this file:
mbed-os\targets\TARGET_STM\TARGET_STM32F4\device\stm32f4xx_hal_msp_template.c

@LMESTM  is there a reason HAL_MspInit/HAL_MspDeInit is defined here or can it be removed to make it possible to override it in the target?

0xc0170, 2017-10-25:
I would use `mbed_sdk_init` but that is for F4 also predefined and invokes some other functions that are mostly related to clocks (SetSysClock, etc). It could provide hooks for F4 targets that might require a bit more setup (board specific).

jeromecoutant, 2017-10-30:
Hi

> @LMESTM is there a reason HAL_MspInit/HAL_MspDeInit is defined here or can it be removed to make it possible to override it in the target?

You are right, this is an issue on our side.
PR is coming soon

closed 2017-12-29 by adbridge