ARMmbed/mbed-os #5368: NUCLEO_F103RB: Operator new[] out of memory on Thread::start

maiorfi, 2017-10-24:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug | Question
- Related issue: `#4570`
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_F103RB

**Toolchain:**
GCC_ARM

**Toolchain version:**
6 2017-q2-update

**mbed-cli version:** 
1.2.2

**mbed-os sha:**
6e0d01cd1 (HEAD, tag: mbed_lib_rev153, tag: mbed-os-5.6.2, tag: latest) Merge pull request #5268 from ARMmbed/release-candidate

**Expected behavior**
Thread starting normally

**Actual behavior**
Operator new[] out of memory in Thread::start (Thread.cpp Line 77)

**Steps to reproduce**
Build following code, main.cpp Line 53 (s_thread_blink_worker.start(...)) causes mbed_die() being called (with consequent abnormal program termination)
----------------------------------------------------------------

```c++
#include "mbed.h"
#include "mbed_stats.h"

static DigitalOut led1(LED1);

static DigitalIn btn(BUTTON1);

static Timer s_timer_1;

Serial pc_serial(SERIAL_TX, SERIAL_RX);

static Thread s_thread_blink_worker;
static EventQueue s_eq_blink_worker;

static Thread s_thread_button_poll;
static EventQueue s_eq_button_poll;

void event_proc_blink_worker(DigitalOut *pled)
{
    pled->write(!pled->read());
}

void event_proc_button_poll()
{
    static int latest_button_state;

    int state = btn.read();

    if (state != latest_button_state)
    {
        pc_serial.printf("[TRACE - %d] Button state : %s\n", s_timer_1.read_ms(), state ? "ON" : "OFF");

        latest_button_state = state;
    }
}

int main()
{
    pc_serial.baud(115200);

    s_timer_1.start();

    mbed_stats_heap_t heap_stats;

    mbed_stats_heap_get(&heap_stats);

    pc_serial.printf("[TRACE - %d] Started\n",s_timer_1.read_ms());

    pc_serial.printf("[TRACE - %d] Current heap: %lu\r\n", s_timer_1.read_ms(), heap_stats.current_size);
    pc_serial.printf("[TRACE - %d] Max heap size: %lu\r\n", s_timer_1.read_ms(), heap_stats.max_size);

    s_eq_blink_worker.call_every(500, event_proc_blink_worker, &led1);
    s_thread_blink_worker.start(callback(&s_eq_blink_worker, &EventQueue::dispatch_forever));

    s_eq_button_poll.call_every(10, event_proc_button_poll);
    s_thread_button_poll.start(callback(&s_eq_button_poll, &EventQueue::dispatch_forever));
}
```

bulislaw, 2017-10-25:
Your board has 20K of ram, you have 3 threads 4K for stack each, plus another 2.5K for system threads that's almost 15k just for stack mem. I'm guessing you are just running out of memory. Try making the stacks smaller: https://github.com/ARMmbed/mbed-os/blob/b83b0b1678002682871fd47e6cfedde386f57591/rtos/Thread.h#L80 you can also overwrite the main thread's stack adding `MAIN_STACK_SIZE` in mbed_app.json

maiorfi, 2017-10-25:
Using following mbed_app.json everything starts correctly:

```json
{
    "config": {
      "thread_stack_size": {
        "value": 1024
      },
      "main_stack_size": {
        "value": 1024
      }
    }
  }
```

Thanks.

closed 2017-10-25 by maiorfi