ARMmbed/mbed-os #5374: NUCLEO_F446RE: CAN RX interrupt ends up in infinite loop

pavels, 2017-10-24:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_F446RE

**Toolchain:**
GCC_ARM

**Toolchain version:**
arm-none-eabi-gcc (Arch Repository) 7.1.0

**mbed-cli version:** 
1.2.0

**mbed-os sha:**
cc7556a92 (HEAD, tag: mbed_lib_rev152, tag: mbed-os-5.6.1, tag: latest, origin/mbed-os-5.6)

---

Something is wrong with the CAN RX interrupt handling.
I am simply attaching my interrupt handler using `attach` method in instance of `CAN` class.
First - it is not possible to call `read()` on the can object in the interrupt routine as there is a mutex lock and this will fail in the interrupt context. On the other end if you do not read from the can interface. The interrupt routine is called again and again in infinite loop so it is not possible to pass the actual read to for example EventQueue as the interrupt handler will be called again and again and the program never leaves the interrupt context.

If i comment out the locking in `CAN.cpp` and execute `read()` in the interrupt handler, all is OK and the interrupt handler is called only when there is some messages left in the mailbox.

0xc0170, 2017-10-24:
cc @bcostm @adustm @LMESTM @jeromecoutant

LMESTM, 2017-10-27:
@0xc0170 can you help describe how this is supposed to work ?

Isn't application supposed to ```read()``` from interrupt handler when an Can Rx interrupt is received ? 
I understand that this is how this works with serial driver for instance and probably why there is no mutex  lock is the serial driver 
```
void SerialBase::lock() {
    // Stub
}
void SerialBase:: unlock() {
    // Stub
}
```

@pavels : maybe an alternative is to disable the interrupt from the interrupt handler before using the EventQueue, then read from your CAN thread and enable interrupt again ...

pavels, 2017-10-27:
@LMESTM if you mean to disable by calling attach with null as first param (handler function) than that is not possible either as there is mutex too. Only option is go directly  to registers - but that sounds like really bad idea (i am changing registers when there is multiple layers of API which expects that nobody changes them directly).

LMESTM, 2017-10-27:
@0xc0170  - can you have a look at the discussion and provide guidance ? I don't see a good way frowared as of now. 

@pavels 
Ok - serial driver and classes is more complex. What I want to say that lock management is clearly different and it is allowed to read serial from the interrupt.

0xc0170, 2017-10-27:
I am looking at this now, I believe that using mutex is an issue here. I'll have a closer look now and send a patch.

LMESTM, 2017-11-06:
@0xc0170 bump ...

0xc0170, 2018-01-24:
Back to this one.

CAN is thread safe by default - It provides virtual locking methods that thus a parent  can override these to handle thread safety on its own.

@c1728p9 @SenRamakri suggestions to resolve this one? Should we make CAN interrupt safe?

jorisa, 2018-01-24:
Bump! We are seeing the same issue...

SenRamakri, 2018-01-25:
We should not get rid of locking in CAN driver, its good to have the locking if multiple threads are to operate on the same object. A good pattern would be to have a event/mailbox trigger a reader thread from the callback and have that thread read the CAN frame/data.

jorisa, 2018-01-25:
@SenRamakri yes but the problem here is that the interrupt only clears after the data has been read. If you do not read the data in the IRQ context, the interrupt will keep firing. Currently it is not possible to read it from the irq context, so there is no way to use the rx interrupt.

pavels, 2018-02-01:
The Serial ended up in the exact same situation so this is relevant https://github.com/ARMmbed/mbed-os/pull/5446

LMESTM, 2018-04-16:
@0xc0170  shouldn't label be updated ? This does not seem to be st specific and maybe the wrong labeling is a possible cause of slow progress here ?

aashishc1988, 2018-10-10:
@pavels I am starting to look at this issue. I understand the underlying problem, but need to reproduce the problem to come up with the right fix. I am new to CAN api, so might be doing something wrong here. I have an example where I am trying to attach my interrupt with CAN, but it seems it is never getting called. 

```
#include "mbed.h"
#include "drivers/CAN.h"
 
Ticker ticker;
DigitalOut led(LED1);
CAN  can(PA_11, PA_12);   // NUCLEO     CAN Rx pin, CAN Tx pin 
char counter = 0;

void printmsg(char *title, CANMessage *msg) {
    printf("%s [%03X]", title, msg->id);
    for(char i = 0; i < msg->len; i++) {
        printf(" %02X", msg->data[i]);
    }
    printf("\n");
}

 
void send() {
    printf("send()\n");
    if(can.write(CANMessage(257, &counter, 1))) {
        counter++;
        printf("Message sent: %d\n", counter);
    } 
    led = !led;
}

void read() {
    CANMessage msg;
    printf("rx()\n");
    if(can.read(msg)) {
        printmsg("Rx message:", &msg);
        led = !led;
    }
}
 
int main() {
    printf("main()\n");
    led = 1;
    can.attach(&read,CAN::RxIrq);
    ticker.attach(&send, 1);
    while(1) {
        printf("loop()\n");
        wait(1);
    }
}
```
The _read_ function never gets called. Can you post your example or let me know if you see something obvious here?

pavels, 2018-10-10:
@aashishc1988 ehm?!

first this will end up with printf in an actual interrupt routine https://os.mbed.com/docs/latest/tutorials/debugging-using-printf-statements.html#printf-from-an-interrupt-context

second - CAN is not that simple - if you want to receive your own messages, you need to enable some kind of test mode https://os.mbed.com/docs/v5.10/mbed-os-api-doxy/classmbed_1_1_c_a_n.html#aaec59a2dc3025d140b88f677893f9dcc

This particular case (mutex / interrupt clash) is the same for basically any peripheral, for which it makes sense to have buffered and unbuffered modes of operation - it is the same thing as Serial vs RawSerial.
Also maybe somebody from MBED team should think about how this is going to be used - for me the most useful thing is to actually have proper buffered CAN interface (because only thing that makes sense to do in the CAN receive interrupt is to stick the massage ASAP into some kind of buffer, signal that there should be processing done afterwards and return)

pavels, 2018-10-10:
By the way, my solution to this particular problem is simple and suits me now i think:

```C
#include "mbed.h"
#ifndef RAW_CAN_H
#define RAW_CAN_H

class RawCAN : public CAN {

public:
    RawCAN (PinName rd, PinName td):CAN(rd, td){};
    RawCAN (PinName rd, PinName td, int hz):CAN(rd, td, hz){};

protected:
    // STUBS
    virtual void lock() {}
    virtual void unlock() {}
};

#endif // RAW_CAN_H
```

aashishc1988, 2018-10-10:
@pavels Thank you for the prompt response. I was not enabling the test mode, I can reproduce the problem now. I will take into account your suggestions for the fix.

aashishc1988, 2018-12-04:
Hi @pavels  , it seems like you have a workaround for now. Another suggestion would be to use mbed-events, https://os.mbed.com/blog/entry/Simplify-your-code-with-mbed-events/, which addresses this problem, but will have more memory footprint. For now, we are not going to investigate this further. Closing this issue. Please feel free to re-open if you have any other questions.

closed 2018-12-05 by 0xc0170