ARMmbed/mbed-os #5380: Nucleo F429ZI does not build with IAR8

teetak01, 2017-10-25:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_F429ZI

**Toolchain:**
IAR

**Toolchain version:**
IAR ANSI C/C++ Compiler V8.20.1.57/LNX for ARM

**mbed-os sha:**
5.6.3

**Expected behavior**
Code compiles

**Actual behavior**

mbed compile -t IAR -m NUCLEO_F429ZI

{code}
00:01:56.716 Compile [  1.8%]: BufferedSerial.cpp
00:01:56.716 
[Error] stm32f4xx_hal_def.h@132,30: [Pe167]: argument of type "unsigned long volatile *" is incompatible with parameter of type "unsigned int const volatile *"
00:01:56.716 [Error] stm32f4xx_hal_def.h@134,34: [Pe167]: argument of type "unsigned long volatile *" is incompatible with parameter of type "unsigned int volatile *"
00:01:56.716 [Error] stm32f4xx_hal_def.h@142,30: [Pe167]: argument of type "unsigned long volatile *" is incompatible with parameter of type "unsigned int const volatile *"
00:01:56.716 [Error] stm32f4xx_hal_def.h@144,0: [Pe167]: argument of type "unsigned long volatile *" is incompatible with parameter of type "unsigned int volatile *"
00:01:57.250 [ERROR] 
00:01:57.250   		newValue = (uint32_t)__LDREXW((volatile unsigned long *)ptr) | mask;
00:01:57.250   		                              ^
00:01:57.250 "app/mbed-os/targets/TARGET_STM/TARGET_STM32F4/device/stm32f4xx_hal_def.h",132  Error[Pe167]: argument of type "unsigned long volatile *" is incompatible with parameter of type "unsigned int const volatile *"
00:01:57.250 
00:01:57.250   	} while (__STREXW(newValue,(volatile unsigned long*) ptr));
00:01:57.250   	                           ^
00:01:57.250 "app/mbed-os/targets/TARGET_STM/TARGET_STM32F4/device/stm32f4xx_hal_def.h",134  Error[Pe167]: argument of type "unsigned long volatile *" is incompatible with parameter of type "unsigned int volatile *"
00:01:57.250 
00:01:57.250   		newValue = (uint32_t)__LDREXW((volatile unsigned long *)ptr) &~mask;
00:01:57.250   		                              ^
00:01:57.250 "app/mbed-os/targets/TARGET_STM/TARGET_STM32F4/device/stm32f4xx_hal_def.h",142  Error[Pe167]: argument of type "unsigned long volatile *" is incompatible with parameter of type "unsigned int const volatile *"
00:01:57.250 
00:01:57.250   	} while (__STREXW(newValue,(volatile unsigned long*) ptr));
00:01:57.250   	                           ^
00:01:57.250 "app/mbed-os/targets/TARGET_STM/TARGET_STM32F4/device/stm32f4xx_hal_def.h",144  Error[Pe167]: argument of type "unsigned long volatile *" is incompatible with parameter of type "unsigned int volatile *"
{code}

0xc0170, 2017-10-25:
cc @deepikabhavnani This was reported in IAR8 PR I believe, the same report, but for ODIN device.

This looks like STM32F4 problem in the hal.

cc @ARMmbed/team-st

deepikabhavnani, 2017-10-25:
__LDREXW, __STREXW and many other defines are update in 8.2 version "arm/inc/c/cmsis_iar.h"

FYI, Pull request for IAR8 fixes should be against branch `feature-iar8`, CI testing for IAR8.2 will be enabled for only feature branch.

deepikabhavnani, 2017-10-26:
Failing Targets: B96B_F446VE, DISCO_F413ZH, DISCO_F429ZI, DISCO_F469NI,  DISCO_F746NG, DISCO_F769NI, DISCO_L475VG_IOT01A, DISCO_L476VG

http://mbed-os.s3-website-eu-west-1.amazonaws.com/?prefix=builds/5329/FAIL/

0xc0170, 2017-10-26:
> __LDREXW, __STREXW and many other defines are update in 8.2 version "arm/inc/c/cmsis_iar.h"

Can you include those definitions here for comparision (v7 vs v8) ? The latest cmsis5 versions have this cmsis_iar already included, see here https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/Core/Include/cmsis_iccarm.h (does this headerfile match regarding this macros update) ? Not certain which one is valid for these targets (either intrisict from iar built in or implementation provided there) ?

Is this bug with ST or in IAR header file (they should provide backward compatible macros, or have a strong reason to break this).

@JonatanAntoni might know more details

deepikabhavnani, 2017-10-26:
Definition in IAR7.8
```
static uint32_t __LDREXW(volatile uint32_t *addr)
{
  __ASM("ldrex r0, [r0]");
}
```
Replaced in IAR8.2 with builtin define
`#define __LDREXW  __iar_builtin_LDREX`

deepikabhavnani, 2017-10-30:
Added fix commit d6da5ca197909445285cf392ee6b4002aa1482aa in https://github.com/ARMmbed/mbed-os/pull/5329

JonatanAntoni, 2017-11-06:
Hi all,

please note that there is a compiler specific version of cmsis_iar.h shipped with the IAR installation. I assume this one

cmonr, 2018-06-28:
Closing this since IAR 8 is not currently supported.

closed 2018-06-28 by cmonr

JonatanAntoni, 2018-07-05:
Just a remark: IAR 8 is the latest release which is under active support. With IAR 7.8 you cannot presume to get anything fixed. Nevertheless, as stated above, please raise issues at CMSIS and we'll check what we can do to keep the compatibility as best as we can.