ARMmbed/mbed-os #5382: Tracking issue for IAR 8 library updates

theotherjimmy, 2017-10-25:
Each library in the following list must be re-compiled with IAR EWARM 8.20 to support the upcoming Mbed OS release 5.7.0 and onward. Please make pull requests containing the updated libraries to the branch `feature-iar8`. 

These are the outstanding libraries:
 - [ ] targets/TARGET_ARM_SSG/TARGET_BEETLE/TOOLCHAIN_IAR/libcordio.0.0.2.a
 - [ ] targets/TARGET_ARM_SSG/TARGET_BEETLE/TOOLCHAIN_IAR/libcordio_platform.0.0.2.a
 - [ ] targets/TARGET_Maxim/TARGET_MAX32610/TOOLCHAIN_IAR/libexactLE.a
 - [ ] targets/TARGET_Maxim/TARGET_MAX32620/TOOLCHAIN_IAR/libexactLE.a
 - [ ] targets/TARGET_Maxim/TARGET_MAX32630/TOOLCHAIN_IAR/libexactLE.a
 - [ ] targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_UBLOX_EVK_ODIN_W2/sdk/TOOLCHAIN_IAR/libublox-odin-w2-driver.a
 - [ ] targets/TARGET_Silicon_Labs/TARGET_SL_RAIL/efr32-rf-driver/rail/TOOLCHAIN_IAR/TARGET_EFR32MG1/librail_efr32xg1_gcc_release.a
 - [ ] targets/TARGET_Silicon_Labs/TARGET_SL_RAIL/efr32-rf-driver/rail/TOOLCHAIN_IAR/TARGET_EFR32MG12/librail_efr32xg12_iar_release.a
 - [ ] targets/TARGET_Realtek/TARGET_AMEBA/TARGET_RTL8195A/device/TOOLCHAIN_IAR/lib_peripheral_mbed_iar.a
 - [ ] targets/TARGET_Realtek/TARGET_AMEBA/TARGET_RTL8195A/device/TOOLCHAIN_IAR/lib_wlan_mbed_iar.a
 - [ ] targets/TARGET_NXP/TARGET_MCUXpresso_MCUS/TARGET_LPC54114/device/TARGET_LPC54114_M4/TOOLCHAIN_IAR/libpower.a
 - [ ] targets/TARGET_NXP/TARGET_MCUXpresso_MCUS/TARGET_LPC546XX/device/TOOLCHAIN_IAR/lib_power.a

@maclobdell @screamerbg  Could you tag the affected parties?
@deepikabhavnani FYI, PRs to your branch incoming.

theotherjimmy, 2017-10-25:
@SeppoTakalo We would also like to see a new nanostack build for IAR 8.20 by 5.7.0.

SeppoTakalo, 2017-10-30:
CC: @mikter @artokin

artokin, 2017-11-02:
IAR8.2 compiled nanostack binaries available https://github.com/ARMmbed/mbed-os/pull/5402.

stevew817, 2017-11-27:
@theotherjimmy Please see #5579 and let me know whether this is appropriate, or whether there should be a separate PR to `feature-iar8`. I'd assume not, since you should be able to rebase on master when my PR gets in, right?

theotherjimmy, 2017-11-27:
@stevew817 Thanks for the update. Does that library work with both IAR v 7.8 and 8.2? if so, then we can rebase `feature-iar8` and check off all of the rail libraries.

0xc0170, 2018-05-22:
@ARMmbed/mbed-os-tools This has been idle for a while, any update for these issues?

cmonr, 2018-06-28:
Closing since IAR 8 is not currently supported. Dealing with multiple compiled binaries for different compiler versions is currently being worked on.

closed 2018-06-28 by cmonr