ARMmbed/mbed-os #5383: Tracking issue for Arm Compiler 6 library updates

theotherjimmy, 2017-10-25:
Each library in the following list must be recompiled with Arm Compiler 6 to support the upcoming Mbed OS release 5.7.0 and onward. Please make pull requests to the branch `feature-armc6` containing the updated libraries and a change that updates the `"supported_toolchains"` targets.json key from `ARM` to `ARMC6`.

These are the outstanding libraries:
 - [ ] targets/TARGET_ARM_SSG/TARGET_BEETLE/TOOLCHAIN_ARM_STD/libcordio.0.0.2.ar
 - [ ] targets/TARGET_ARM_SSG/TARGET_BEETLE/TOOLCHAIN_ARM_STD/libcordio_platform.0.0.2.ar
 - [ ] targets/TARGET_Maxim/TARGET_MAX32610/TOOLCHAIN_ARM_STD/libexactLE.ar
 - [ ] targets/TARGET_Maxim/TARGET_MAX32620/TOOLCHAIN_ARM_STD/libexactLE.ar
 - [ ] targets/TARGET_Maxim/TARGET_MAX32630/TOOLCHAIN_ARM_STD/libexactLE.ar
 - [ ] targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_UBLOX_EVK_ODIN_W2/sdk/TOOLCHAIN_ARM/libublox-odin-w2-driver.ar
 - [ ] targets/TARGET_Realtek/TARGET_AMEBA/TARGET_RTL8195A/device/TOOLCHAIN_ARM_STD/lib_peripheral_mbed_arm.ar
 - [ ] targets/TARGET_Realtek/TARGET_AMEBA/TARGET_RTL8195A/device/TOOLCHAIN_ARM_STD/lib_wlan_mbed_arm.ar
 - [ ] targets/TARGET_NXP/TARGET_MCUXpresso_MCUS/TARGET_LPC54114/device/TARGET_LPC54114_M4/TOOLCHAIN_ARM_STD/libpower.ar
 - [ ] targets/TARGET_NXP/TARGET_MCUXpresso_MCUS/TARGET_LPC546XX/device/TOOLCHAIN_ARM_STD/lib_power.ar

@screamerbg @maclobdell

theotherjimmy, 2017-10-25:
@SeppoTakalo We would also like to see a new nanostack build for Arm Compiler 6 for 5.7.0.

artokin, 2017-10-25:
@theotherjimmy , PR https://github.com/ARMmbed/mbed-os/pull/5381 contains nanostack build with ARMC6.

SeppoTakalo, 2017-10-30:
CC: @mikter @artokin

theotherjimmy, 2017-10-30:
@pan- Do you think we could move that cordio stack and build it ourselves?

artokin, 2017-11-02:
ARMC6 binaries available in https://github.com/ARMmbed/mbed-os/pull/5401.

pan-, 2017-11-09:
@theotherjimmy The cordio stack in `TARGET_CORDIO` is built by our CI; I need to modify the buildscript to use ARMC6 instead of ARMC5. However we cannot generate cordio builds for the BEETLE or MAXIM's targets.

theotherjimmy, 2017-11-09:
@pan- I will have to turn BLE off for those targets then.

theotherjimmy, 2017-11-09:
@pan- After #5406, you could just configure both `MBED_ARM_PATH` for ARMC5 and `MBED_ARMC6_PATH` for ARMC6, and the tools will compile with whichever arm compiler is correct for that branch.

0xc0170, 2018-05-22:
@ARMmbed/mbed-os-tools This has been idle for a while, any update for these issues?

deepikabhavnani, 2018-12-28:
Closing this, @SenRamakri - Please create new one, if library updates are needed for ARMC6 as well.

closed 2018-12-28 by deepikabhavnani