ARMmbed/mbed-os #5386: malloc unexpected behaviour on GCC_ARM compiler

maciejbocianski, 2017-10-26:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
all

**Toolchain:**
GCC_ARM

**Toolchain version:**
gcc version 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437] (GNU Tools for ARM Embedded Processors 6-2017-q2-update)

**mbed-cli version:** 
1.2.0

**mbed-os sha:**
38ba693c5

**Expected behavior**
When try to allocate more memory than heap can handle we get NULL from malloc and further allocations (within heap size) work fine

**Actual behavior**
When try to allocate more memory than heap can handle we get NULL from malloc and any further allocation (within heap size) return NULL

**Steps to reproduce**

Compile and run below code
```
extern unsigned char *mbed_heap_start;
extern uint32_t mbed_heap_size;
uint32_t max_mbed_heap_size = mbed_heap_size;

int main()
{
	void *data = NULL;

	while(true)
	{
		data = malloc(max_mbed_heap_size);
		if(data || max_mbed_heap_size == 0) {
			break;
		} else {
			max_mbed_heap_size--;
		}
	}

	printf("max_mbed_heap_size: %lu \r\n", max_mbed_heap_size);
	printf("mbed_heap_start: %X \r\n", mbed_heap_start);
	printf("mbed_heap_size: %lu \r\n", mbed_heap_size);
	printf("data: %p \r\n\r\n", data);
}
```

Output from NUCLEO_F070RB
```

GCC_ARM
max_mbed_heap_size: 0
mbed_heap_start: 2000275C
mbed_heap_size: 5284
data: 0x0

ARM
max_mbed_heap_size: 6556
mbed_heap_start: 2000224C
mbed_heap_size: 6580
data: 20002260

IAR
max_mbed_heap_size: 4044
mbed_heap_start: 200006E8
mbed_heap_size: 4096
data: 200006f0
```

R1oT01, 2017-10-27:
On custom Board with STM32F072VB - same toolchain but on Eclipse:
Is this relevant?

`max_mbed_heap_size: 0 <\r><\n>
mbed_heap_start: 20002680 <\r><\n>
mbed_heap_size: 5504 <\r><\n>
data: 0x0`

Best regards
Chris

0xc0170, 2017-10-27:
@maciejbocianski we implement `sbrk` for gcc in the retarget code file.  What `sbrk` returns if you pass 0 (returns the current program break value) before the failing and after?

cc @bulislaw @c1728p9

maciejbocianski, 2017-10-30:
@0xc0170 @bulislaw @c1728p9 
Regarding our friday conversation
I did some test with malloc and _sbrk

Below output from "Test program 2" (program code in issue description)

We can see that on `main` enter there is 1140B already  requested by _sbrk
During the test we allocate  10x100B and then an 11th loop iteration `malloc` request another memory `_sbrk_incr: 4096`
and this request fails since `new_heap(20004000) >= __get_MSP(20003FC8)`

```
*** main begin ***

_sbrk_incr: 1140
_sbrk__end__: 521
prev_heap: 20002B8C
new_heap: 20003000

*** test begin ***
allocated: 100
allocated total: 100

allocated: 100
allocated total: 200

allocated: 100
allocated total: 300

allocated: 100
allocated total: 400

allocated: 100
allocated total: 500

allocated: 100
allocated total: 600

allocated: 100
allocated total: 700

allocated: 100
allocated total: 800

allocated: 100
allocated total: 900

allocated: 100
allocated total: 1000

_sbrk_incr: 4096
_sbrk__end__: 521
prev_heap: 20003000
new_heap: 20004000
_sbrk_ret_1: 20004000
*** test end ***

max_mbed_heap_size: 1000
mbed_heap_start: 20002774
mbed_heap_size: 5260
data: 0x20002f30
__get_MSP(): 20003FC8
*** main end ***

```

c1728p9, 2017-10-30:
> The question is if we can force malloc to request memory in smaller chunks than 4kB at least for platforms with small RAM?

The behavior I saw when I looked at this a while back was that calls to `_sbrk` make requests up to a 4k boundary.  In the above case:
```
0x20002B8C + 1140-> 0x20003000
0x20003000 + 0x1000 -> 0x20004000
```

If the interrupt stack was moved to come before the heap then the end of the heap would be on a 4k boundary, and in theory you would make use of the whole heap. I'm not sure what it would take to make this change though.

maciejbocianski, 2017-10-31:
other question is why first allocation likely performed by `printf` is not 4kB but 1140B?

c1728p9, 2017-10-31:
@maciejbocianski the first allocation of 1140B brings the heap limit to a 4k boundary ```0x20002B8C + 1140 -> 0x20003000```.

maciejbocianski, 2017-11-03:
Looks like by this 4kB alignment we lose 3kB of heap, what is huge problem for platforms with small RAM

jeromecoutant, 2017-11-15:
Any update about this issue ?
Thx

bulislaw, 2017-11-15:
Right now I don't think we have a path forward, more investigation is needed.

cmonr, 2017-12-07:
@bulislaw @jeromecoutant @maciejbocianski @c1728p9 Has anyone continued to look at this? Is this still a major bug?

adamgreen, 2017-12-07:
You could upgrade to GCC ARM Embedded 6-2017-q1-update or newer and link with newlib-nano instead of the full newlib Standard C Library. That version of newlib-nano synchronizes multithreaded access to the heap. newlib-nano doesn't do this 4k paging as it is the version of newlib meant for small memory devices like this.

jeromecoutant, 2017-12-13:
> That version of newlib-nano synchronizes multithreaded access

This means that the restriction to forbid thread with nano gcc lib doesn't exist anymore ?
See #4885

bulislaw, 2017-12-18:
@cmonr I have a task on me to look into it.

c1728p9, 2018-02-21:
Newlib nano doesn't have thread safe file access so even with a protected heap, it may still be unsafe to use.

deepikabhavnani, 2018-10-05:
Malloc unaligned behavior for GCC_ARM cannot be fixed. GCC_ARM toolchain expects heap to be 4K aligned. If heap is not aligned at 4K boundary allocation will fail. 
This issue might be considered during major release, but for now it cannot be fixed

closed 2018-10-05 by deepikabhavnani

bmcdonnell-ionx, 2018-10-05:
@deepikabhavnani, @adbridge,

> This issue might be considered during major release

Are you still tracking that somehow, since this issue is now closed?

deepikabhavnani, 2018-10-05:
Yes it will be tracked internally