ARMmbed/mbed-os #5387: Mbed test linker Errors - IAR8

deepikabhavnani, 2017-10-26:
Mbed test build fails on IAR8 feature branch with below error, refer PR #5329 for more information:
```

        Link: heap_and_stack
        [DEBUG] Link: /usr/local/iccarm-8.20.1.57/bin/ilinkarm -f /builds/ws/mbed-os-build-matrix/341/KL25Z_IAR8/sources/mbed-os/BUILD/tests/KL25Z/IAR/./TESTS/mbedmicro-rtos-mbed/heap_and_stack/.link_files.txt
        [DEBUG] Return: 2
        [DEBUG] Output: 
        [DEBUG] Output:    IAR ELF Linker V8.20.1.57/LNX for ARM
        [DEBUG] Output:    Copyright 2007-2017 IAR Systems AB.
        [DEBUG] Output: 
        [DEBUG] Output:   32 576 bytes of readonly  code memory
        [DEBUG] Output:    1 880 bytes of readonly  data memory
        [DEBUG] Output:   16 243 bytes of readwrite data memory
        [DEBUG] Output: 
        [DEBUG] Output: Errors: 1
        [DEBUG] Output: Warnings: none
        [DEBUG] Output: 
        [DEBUG] Output: Link time:   0.17 (CPU)   1.00 (elapsed)
        [DEBUG] Errors: Error[Lp011]: section placement failed
        [DEBUG] Errors:             unable to place sections/blocks with a total estimated minimum size
        [DEBUG] Errors:                       of 0xf4 bytes (the smallest of which is Section
        [DEBUG] Errors:                       .bss.os.msgqueue.cb (rtx_lib.o #11) (min size 0x34, align
        [DEBUG] Errors:                       0x4) in: <[0x000000c0-0x000003ff],
        [DEBUG] Errors:                       [0x00008920-0x0001ffff], [0x1ffff31c-0x1ffff31f],
        [DEBUG] Errors:                       [0x200029c0-0x20002fff]>
```
### Description
- Type: Bug 
- Priority: Blocker 

**Target**
KL25Z and KL46Z

**Toolchain:**
IAR8.2

0xc0170, 2017-10-26:
Is this included also with dynamic heap support for IAR8 ? I assume not.

deepikabhavnani, 2017-10-26:
It is with dynamic heap support. Got the issue on dynamic heap PR

deepikabhavnani, 2017-10-31:
Splited RAM section not compatible with dynamic HEAP. RAM define/section if splited, IAR expanding size considers just one section while heap creation which results in smaller chunk for heap. 
Fixed in commit 8e571079e66f0f3ee385d936eda987dad709715b

deepikabhavnani, 2017-10-31:
0xc0170 - Please add Freescale team for review

0xc0170, 2017-11-10:
> Fixed in commit 8e57107

Is it functional ? I recall when we had one region (some time ago), allocation need to be aware about boundaries (no object should cross the boundary that is set to 0x2000 0000 as I recall. Limitation of their controller that does not support access independently to both RAM banks at the same time). I can not find the issue now for reference :/ That fix is the same for ARMCC and GCC, one RAM region?

@mmahadevan108 @maclobdell  Can you please have a look at this issue.

deepikabhavnani, 2017-11-10:
As per @c1728p9, crossing boundaries should not be an issue with M0 devices. I am not sure of any controller limitation, hence would be best to have review from Freescale.

Also we did the `morph test` on https://github.com/ARMmbed/mbed-os/pull/5329, after fix but we do not have any KL series device in CI testing

mmahadevan108, 2017-11-10:
Looks good to me.

deepikabhavnani, 2018-06-01:
This issue was resolved by IAR in upcoming release.

closed 2018-06-01 by deepikabhavnani