ARMmbed/mbed-os #5388: Transplantation of mbed os

zhanghongyuanvg, 2017-10-27:
Hi.
I want to transplant the mbed operating system to the nucleo_f072rb development board under Linux. What should I do?

0xc0170, 2017-10-27:
> I want to transplant the mbed operating system to the nucleo_f072rb development board under Linux. What should I do?

Can you please provide more details. What are you having problems with? Is it mbed-os development under linux?

Is there something missing in our documentation pages: https://docs.mbed.com ?

zhanghongyuanvg, 2017-10-28:
Thanks for your reply.
Just like the mini2440 development board was transplanted linux operating system, I would like to cut mbed operating system to compile a mirror file, and then transplanted to the nucleo_f072rb development board to run it.

amq, 2017-10-28:
@zhanghongyuanvg if I understand you correctly, you need NUCLEO_F072RB to be supported by mbed-os. If this is correct, then I have good news for you: it's already supported.

zhanghongyuanvg, 2017-10-29:
@amq Thanks.I will try to do it.

bulislaw, 2017-10-30:
Please reopen this issue if you require more assistance.

closed 2017-10-30 by bulislaw

jacob-larsen-samsung, 2019-07-09:
git clone https://github.com/ARMmbed/mbed-os.git
git clone https://github.com/ARMmbed/mbed-os-example-blinky.git

cd mbed-os-example-blinky

Contents of .mbed:
ROOT=.
TOOLCHAIN=GCC_ARM
TARGET=K64F
MBED_OS_DIR=../mbed-os

$ mbed compile -v
[mbed] Working path "/mbed" (program)
[mbed] WARNING: Missing Python modules were not auto-installed.
       The Mbed OS tools in this program require the following Python modules: hidapi, pywin32, wmi, psutil
       You can install all missing modules by running "pip install -r requirements.txt" in "../mbed-os"
       On Posix systems (Linux, etc) you might have to switch to superuser account or use "sudo"
---
[mbed] Exec "/usr/bin/python -u /mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build ./BUILD/K64F/GCC_ARM -v" in "/mbed"
Building project mbed (K64F, GCC_ARM)
Scan: mbed
Traceback (most recent call last):
  File "/mbed-os/tools/make.py", line 78, in wrapped_build_project
    *args, **kwargs
  File "/mbed-os/tools/build_api.py", line 584, in build_project
    src_paths, toolchain, inc_dirs=inc_dirs)
  File "/mbed-os/tools/resources/__init__.py", line 582, in scan_with_toolchain
    toolchain.config.load_resources(self)
  File "/mbed-os/tools/config/__init__.py", line 1514, in load_resources
    self.validate_config()
  File "/mbed-os/tools/config/__init__.py", line 1457, in validate_config
    raise error
UndefinedParameter: Attempt to override undefined parameter 'platform.stack-stats-enabled' in 'application[*]'
[mbed] ERROR: "/usr/bin/python" returned error.
       Code: 1
       Path: "/mbed"
       Command: "/usr/bin/python -u /mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build ./BUILD/K64F/GCC_ARM -v"
       Tip: You could retry the last command with "-v" flag for verbose output

jacob-larsen-samsung, 2019-07-09:
This workaround makes compile work:

ln -s ../mbed-os mbed-os

0xc0170, 2019-07-09:
@jacob-larsen-samsung I believe this is the wrong issue (not related to your report). You created earlier a new one, please follow up there.

jacob-larsen-samsung, 2019-07-09:
Sorry, hit the wrong tab