ARMmbed/mbed-os #5404: uninitialized section is not supported by all the platforms

alzix, 2017-10-30:
I wish to place some variables in an uninitialized section in RAM. 
For me it is only required for testing purpose.
By examining linker scripts for GCC_ARM, ARM & IAR for supported platforms, I see that the section is sometimes either missing or named differently

### Description
- Type: Enhancement
- Related issue: `N/A
- Priority: 

---------------------------------------------------------------
## Bug

**Target**
ALL

**Toolchain:**
GCC_ARM  & ARM & IAR

**Toolchain version:**
N/A
**mbed-cli version:** 
`N/A`

**mbed-os sha:**
HEAD

**DAPLink version:**
N/A
**Expected behavior**

**Actual behavior**

**Steps to reproduce**

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**
code relying on uninitialized section is not portable between platforms

**Suggested enhancement**
update linker scripts by adding uninitialized section to all the platforms

**Pros**
better SW portability

**Cons**
N/A
-----------------------------------------------------------------

## Question

**How to?**

0xc0170, 2017-11-03:
> By examining linker scripts for GCC_ARM, ARM & IAR for supported platforms, I see that the section is sometimes either missing or named differently

Can you provide some examples how it is named in some linker script files?

cc @deepikabhavnani @c1728p9

alzix, 2017-11-05:
for examples please take a look at:

- https://github.com/ARMmbed/mbed-os/blob/f18d2d862abbd44f6f77fe49464bf45253c08e90/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF51822_UNIFIED/device/TOOLCHAIN_ARM_STD/TARGET_MCU_NORDIC_32K/nRF51822.sct#L22

- https://github.com/ARMmbed/mbed-os/blob/f59b5f8b75223937137240da0fbc3556de318ec3/targets/TARGET_NXP/TARGET_LPC408X/device/TOOLCHAIN_GCC_CR/LPC407x_8x.ld#L174

- https://github.com/ARMmbed/mbed-os/blob/6912a9dc0d06c1296483945a7f4a0c363f4d8bb1/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device/TOOLCHAIN_GCC_ARM/MK64FN1M0xxx12.ld#L317

Example for platform where uninitialized section is not supported at all:
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NXP/TARGET_LPC82X/TARGET_SSCI824/device/TOOLCHAIN_GCC_ARM/LPC824.ld

SenRamakri, 2018-12-06:
@alzix - Currently, if you want to use an uninitialized section, please create a separate section/region with appropriate linker flags(depending on toolchain). But moving forward, we are looking into the possibility of tools to auto-generate the linker scripts for tool-chains which may address your question in a better way. Note the auto-generation of linker-scripts is still an idea but not planned. So, currently the best option is to create new section/region as required for your purpose. Let me know if you have more questions or we can talk directly.

closed 2018-12-06 by SenRamakri