ARMmbed/mbed-os #5414: UBLOX_EVK_ODIN_W2 linking error with IAR8.2

artokin, 2017-11-01:
### Description
- Type: Bug

---------------------------------------------------------------
## Bug

**Target**
UBLOX_EVK_ODIN_W2

**Toolchain:**
IAR

**Toolchain version:**
8.2

**mbed-cli version:** 
(`1.1.1`)

**mbed-os sha:**
8e57107 (HEAD -> feature-iar8, origin/feature-iar8) Splited RAM section not compatible with dynamic HEAP.

**Actual behavior**
Application fails to link with IAR8.2. 

```javascript
Link: mbed-os-example-mesh-minimal
Error[Li060]: module "cb_main.o(libublox-odin-w2-driver.a)" contains C++ code
          built for use with the (E)C++98 library. This code is not link
          compatible with the new (C++14) library.
Error[Li009]: runtime model conflict: Module
          cb_main.o(libublox-odin-w2-driver.a) specifies that '__CPP_Language'
          must be 'C++', but module main.o has the value 'C++14'. Modules
          compiled with C++14 are not link-compatible with modules that are
          compiled with older C++.
Error[Li060]: module "cb_ticker_wrapper.o(libublox-odin-w2-driver.a)" contains
          C++ code built for use with the (E)C++98 library. This code is not
          link compatible with the new (C++14) library.
Warning[Lt009]: Inconsistent wchar_t size
            rf_driver_storage.o(libnanostack.a) and 182 other objects in 108
                      different libraries have wchar_t size 16 bits
            main.o and 555 other objects, some of them in dlpp7M_tl_fc.a and
                      107 other libraries have wchar_t size 32 bits
[ERROR] Error[Li060]: module "cb_main.o(libublox-odin-w2-driver.a)" contains C++ code
          built for use with the (E)C++98 library. This code is not link
          compatible with the new (C++14) library.
Error[Li009]: runtime model conflict: Module
          cb_main.o(libublox-odin-w2-driver.a) specifies that '__CPP_Language'
          must be 'C++', but module main.o has the value 'C++14'. Modules
          compiled with C++14 are not link-compatible with modules that are
          compiled with older C++.
Error[Li060]: module "cb_ticker_wrapper.o(libublox-odin-w2-driver.a)" contains
          C++ code built for use with the (E)C++98 library. This code is not
          link compatible with the new (C++14) library.
Warning[Lt009]: Inconsistent wchar_t size
            rf_driver_storage.o(libnanostack.a) and 182 other objects in 108
                      different libraries have wchar_t size 16 bits
            main.o and 555 other objects, some of them in dlpp7M_tl_fc.a and
                      107 other libraries have wchar_t size 32 bits

[mbed] ERROR: "/usr/bin/python" returned error code 1.
[mbed] ERROR: Command "/usr/bin/python -u /home/arto/github/mbed-os-example-mesh-minimal/mbed-os/tools/make.py -t IAR -m UBLOX_EVK_ODIN_W2 --source . --build out/UBLOX_EVK_ODIN_W2_IAR_ATMEL_6lp/ -c --app-config ./configs/mesh_6lowpan.json" in "/home/arto/github/mbed-os-example-mesh-minimal"
```

**Steps to reproduce**
1. Install IAR 8.2 toolchain
2. checkout mbed-os-example-mesh-minimal
3. mbed deploy
4. change mbed-os revision to feature-iar8
5. compile the application with IAR8.2 "mbed compile --build out/UBLOX_EVK_ODIN_W2_IAR_ATMEL_6lp/ -m UBLOX_EVK_ODIN_W2 -t IAR -c --app-config ./configs/mesh_6lowpan.json"

deepikabhavnani, 2017-11-02:
Error is because of incompatible libraries. Related: https://github.com/ARMmbed/mbed-os/issues/5382

cmonr, 2018-07-03:
@artokin Fyi, IAR 8 is not yet supported in Mbed OS.

SenRamakri, 2018-07-23:
Hi @artokin - As @cmonr noted above IAR8 is not yet supported and is a known issue. We will be adding support for IAR8 in future and should fix this issue. So, closing this as of now with the above said info.

closed 2018-07-23 by SenRamakri

sohaibqamar, 2018-08-08:
Hi,
I'm using IAR Workbench 8.30.1 to Build and Debug the code for mbed NXP LPC 1768.
I have installed only the IAR for ARM and exported the mbed-blinky from online compiler.
I'm trying to build with default options of IAR. I am getting the following errors during Make. Please help.

`Building configuration: mbed_blinky - mbed_blinky
Updating build tree...
main.cpp
Linking
Error[Li060]: module "mbed_retarget.o" contains C++ code built for use with the (E)C++98 library. This code is not link compatible with the new (C++14) library.
Error[Li009]: runtime model conflict: Module mbed_retarget.o specifies that '__CPP_Language' must be 'C++', but module main.o has the value 'C++14'. Modules compiled with C++14 are not link-compatible with modules that are compiled
with older C++.
Error[Li060]: module "mbed_poll.o(mbed.a)" contains C++ code built for use with the (E)C++98 library. This code is not link compatible with the new (C++14) library.
Error[Li060]: module "FilePath.o(mbed.a)" contains C++ code built for use with the (E)C++98 library. This code is not link compatible with the new (C++14) library.
Error[Li060]: module "FileHandle.o(mbed.a)" contains C++ code built for use with the (E)C++98 library. This code is not link compatible with the new (C++14) library.
Error[Li060]: module "Timer.o(mbed.a)" contains C++ code built for use with the (E)C++98 library. This code is not link compatible with the new (C++14) library.
Error[Li060]: module "FileBase.o(mbed.a)" contains C++ code built for use with the (E)C++98 library. This code is not link compatible with the new (C++14) library.
Error[Li005]: no definition for "__aeabi_read_tp" [referenced from errno.o(dl7M_tlf.a)]
Error while running Linker

Total number of errors: 8
Total number of warnings: 1

sg-, 2018-12-18:
hi @sohaibqamar are you using an mbed 2 build (this is identified by a file named mbed.bld in your program)? If so, you'd need to use the mbed-dev source variant or mbed-os.