ARMmbed/mbed-os #5421: build fails for ARM toolchain

gerhardtgao, 2017-11-02:
### Description
- Type: Bug
- Related issue: `#abc`
- Priority: Blocker

---------------------------------------------------------------
## Bug
"--cpu=Cortex-M" is not generated in .link_files.txt

**Target**
K64F|EFM32GG_STK3700| .......

**Toolchain:**
ARM

**Toolchain version:** ARM Compiler 5.06 update 5 (build 528)

**mbed-cli version:** 
(`mbed --version`) 1.2.2

**Expected behavior**
build success

**Actual behavior**
Link: mbed-os-example-blinky
Error: C9912E: No **--cpu selected**
Error: L6636E: Pre-processor step failed for './mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device/TOOLCHAIN_ARM_STD/MK64FN1M0xxx12.sct'
Error: L6372E: Image needs at least one load region.
Not enough information to list the image map.
Finished: 1 information, 0 warning and 2 error messages.
[ERROR] Error: C9912E: No --cpu selected
Error: L6636E: Pre-processor step failed for './mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device/TOOLCHAIN_ARM_STD/MK64FN1M0xxx12.sct'
Error: L6372E: Image needs at least one load region.
Not enough information to list the image map.
Finished: 1 information, 0 warning and 2 error messages.

**Steps to reproduce**
import mbed-os-example-blinky project
mbed comiple -m K64F -t ARM
mbed comiple -m efm32gg_stk3700 -t ARM

** a possible method to fix **
in mbed-os\tools\toolchains\arm.py, at line 79, change 
self.ld = [join(ARM_BIN, "armlink")] + self.flags['ld']
to
self.ld = [join(ARM_BIN, "armlink")] + self.flags['ld'] + ["--cpu=%s" % cpu]

gerhardtgao, 2017-11-02:
@jimmy

theotherjimmy, 2017-11-02:
Thanks for the bug report. The big question on my mind is: why is this not failing for everyone all of the time?

I have exactly the same version of armcc and I can build fat-fs just fine. 
```
Product: ARM Compiler 5.06
Component: ARM Compiler 5.06 update 5 (build 528)
```

and 

```
# mbed compile -t arm -m K64F
Building project fat-fs (K64F, ARM)
Scan: .
[SNIP]
Link: fat-fs
Elf2Bin: fat-fs
+------------------+-------+-------+------+
| Module           | .text | .data | .bss |
+------------------+-------+-------+------+
| [lib]/c_w.l      | 13129 |    16 |  392 |
| [lib]/cpprt_w.l  |    42 |     0 |    0 |
| [lib]/fz_wm.l    |    18 |     0 |    0 |
| [lib]/m_wm.l     |    48 |     0 |    0 |
| anon$$obj.o      |    32 |     0 | 1024 |
| main.o           |  1094 |     0 |  132 |
| mbed-os/drivers  |   130 |     0 |    0 |
| mbed-os/features | 18806 |     4 |  780 |
| mbed-os/hal      |  1424 |     2 |   64 |
| mbed-os/platform |  3172 |    37 |  532 |
| mbed-os/rtos     | 11402 |  2569 | 4252 |
| mbed-os/targets  |  9303 |   104 |  324 |
| Subtotals        | 58600 |  2732 | 7500 |
+------------------+-------+-------+------+
Total Static RAM memory (data + bss): 10232 bytes
Total Flash memory (text + data): 61332 bytes

Image: ./BUILD/K64F/arm/fat-fs.bin
```

Maybe something is up with your environment?

gerhardtgao, 2017-11-02:
hmmmm, then can you see “--cpu=Cortex-M3” in your BUILD\K64F\ARM\.link_files.txt file?

theotherjimmy, 2017-11-02:
Nope.

0xc0170, 2017-11-03:
@gerhardtgao If you look at arm implementation https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/arm.py#L310

you can see that --cpu flag is set there. CAn you use verbose and print entire linker command (assuming based on the output above is a linker that is failing) ?
What about these flags, which one is getting set in your case?

What is your mbed-os version?

theotherjimmy, 2017-11-03:
@0xc0170 That's for ARMC6. Please don't confuse the compiler versions.

0xc0170, 2017-11-03:
> @0xc0170 That's for ARMC6. Please don't confuse the compiler versions.

Not intentionally ! :-) ` --cpu` is present for common flags in ARMC5. I updated my comment above.

theotherjimmy, 2017-11-03:
@0xc0170 Common flags are not used by the linker, so your comment is still incorrect.

gerhardtgao, 2017-11-06:
In <ARM Compiler armlink User Guide>, it is said that, "armlink assumes --cpu=ARM7TDMI if you do not specify a --cpu option." So I don't know if --cpu should appear in armlinks parameter list.

theotherjimmy, 2017-11-06:
Thanks for the tip @gerhardtgao Do you have a link?

gerhardtgao, 2017-11-06:
http://infocenter.arm.com/help/topic/com.arm.doc.dui0377g/DUI0377G_mdk_armlink_user_guide.pdf

theotherjimmy, 2017-11-06:
Thanks section 9.26 is where the `--cpu` flag's default is defined.

gerhardtgao, 2017-11-06:
but for k64， --cpu=cortex-m3 should appear in .link_files.txt, default vaule is not suitable.

theotherjimmy, 2017-11-06:
That's true. 

I think that the linker may not need it unless LTO is enabled, as it's directly tied to code generation. My understanding is that the missing `--cpu` argument does not harm non-LTO bulids.

You are correct in pointing out that it should be passed to the linker. It will be resolved when we switch to Arm Compiler 6, as the `--cpu` flag is passed by the Arm Compiler 6 implementation.

theotherjimmy, 2017-11-06:
Also, for a K64F, `--cpu=cortex-m4` is the correct option.

0xc0170, 2017-11-14:
@gerhardtgao We have not been able to reproduce this issue, is this still relevant? You are still having problems to build a project?

gerhardtgao, 2017-12-04:
I forked the repo and did some modification in my repo.
in mbed-os\tools\toolchains\arm.py, at line 79, change
self.ld = [join(ARM_BIN, "armlink")] + self.flags['ld']
to
self.ld = [join(ARM_BIN, "armlink")] + self.flags['ld'] + ["--cpu=%s" % cpu]

0xc0170, 2018-06-19:
Was this fixed by #5421 ?