ARMmbed/mbed-os #5423: EFM32: EM1 blocking with wait

chrissnow, 2017-11-02:
### Description
- Type: Bug 
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
EFM32PG - probably all EFM32
I know we should use Thread::wait() where possible but wait is still required, especially for sub ms timing.

**Expected behavior**
wait should not lock EM1 forever, Prior to return it should clear the lock.
**Actual behavior**
 A call to wait will block EM1 and will prevent sleeping later on.

**Steps to reproduce**
Call wait_us(),wait_ms(),wait() and look at sleep_block_counter before after.

0xc0170, 2017-11-03:
cc @stevew817 @akselsm

stevew817, 2017-11-24:
This is a sticky one. Wait exercises the high-precision counter, and initializes it. At that point, the API's expectation is that the underlying timer keeps running to keep calls to read_time actually returning elapsed time.
Until we get reference counting on the ticker/timer usage, and explicitly turn it off when not in use, we can't just enable EM2 again, since that would stop the clock to the high-precision timer.

0xc0170, 2017-11-24:
>  Until we get reference counting on the ticker/timer usage, and explicitly turn it off when not in use, we can't just enable EM2 again, since that would stop the clock to the high-precision timer.

This branch should help - feature-hal-spec-sleep - defines that deepsleep turns off high speed clock.
All C++ time based objects have sleep manager in place, thus if not used , deepsleep should be allowed (in this case EM2 would be enabled or lower). Will that address this issue?

PR #5581 (2017-12-12): Remove custom Silicon Labs sleep management
## Description

Standardize on the mbed sleep manager by removing the sleepmodes API, and statically redirecting hal_sleep to EM1 and hal_deepsleep to EM2. This should resolve #5423, #5005 and #4082.

@0xc0170 @bulislaw 

## Status

**READY**

## Migrations

If users were calling blockSleepmode and unblockSleepmode:

- (un)blockSleepmode(EM0) and (un)blockSleepmode(EM1): don't call anymore
- (un)blockSleepmode(EM2) and below: replace with sleep_manager_(un)lock_deep_sleep() from mbed_sleep.h
closed 2017-12-12 by 0xc0170