ARMmbed/mbed-os #5439: Flash programming (write) fails

kylestein-arm, 2017-11-06:
### Description
- Type: Bug
- Priority: Major 

The write of a FOTA image to internal flash results in incorrect data being written to internal flash on Odin platforms.  This results in a failed hash compare using the application
https://github.com/ARMmbed/fota-demo/commit/aefc7b93fa904d98fd438873c7293b565e5c2eff

Some debug has been done on the issue (details below).
Down-revving the flash driver per 
https://github.com/ARMmbed/mbed-bootloader-internal/pull/41
seems to eliminate the issue.

---------------------------------------------------------------
## Bug

**Target**
UBLOX_EVK_ODIN_W2

**Toolchain:**
GCC_ARM

**mbed-cli version:** 
1.2.2

**mbed-os sha:**
6e0d01cd1 (tag: mbed_lib_rev153, tag: mbed-os-5.6.2, tag: latest) Merge pull request #5268 from ARMmbed/release-candidate

**Expected behavior**
Data read from flash matches the source dat used to program (write to) flash.

**Actual behavior**
flash.program() returns a success return code but on a subsequent read, the data (at the same address) does not match the data provided on the write.

Possible relevant details from debug:

- data in this instance was being written in 1024 byte chunks.  A large majority of writes were successful
- The first unsuccessful write occurred at flash address 0x08100000 which corresponds to the start of the second bank of flash.  This is followed by both successful and unsuccessful writes.
- The failure is consistent with the same incorrect data being written to the same addresses on each successive run.  (The SHA of the incorrect image is consistent from run to run)
- The following code snippet and log output illustrates the issue (note this code was added for debug and is not present in the commit referenced above.)

```
bool writeActiveFirmware(uint32_t index, arm_uc_firmware_details_t* details)
{
    tr_debug("writeActiveFirmware");

    bool result = false;

    fota::printProgress(0, 100, "Write FW");
    if (details)
    {
        const uint32_t pageSize = flash.get_page_size();
        const uint32_t chunkSize = get_write_chunk_size();

        /* we require app_start_addr fall on a page size boundary */
        uint32_t app_start_addr = MBED_CONF_APP_FIRMWARE_METADATA_HEADER_ADDRESS
                                + MBED_CONF_APP_FIRMWARE_METADATA_HEADER_SIZE;

        printf("XXX - writing image to appstartaddr %" PRIX32 "\n", app_start_addr);

        ASSERT((app_start_addr % pageSize) == 0,
               "Application (0x%" PRIX32 ") does not start on a "
               "page size (0x%" PRIX32 ") aligned address\r\n",
               app_start_addr,
               pageSize);

        /* round down the read size to a multiple of the page size
           that still fits inside the main buffer.
        */
        uint32_t readSize = (BUFFER_SIZE / chunkSize) * chunkSize;

        arm_uc_buffer_t buffer = {
            .size_max = readSize,
            .size     = 0,
            .ptr      = buffer_array
        };

        int retval = 0;
        uint32_t offset = 0;

        /* write firmware */
        while ((offset < details->size) &&
               (retval == 0))
        {
            /* clear most recent UCP event */
            event_callback = CLEAR_EVENT;

            /* fill buffer using UCP */
            arm_uc_error_t ucp_status = ARM_UCP_Read(index, offset, &buffer);
            printf("XXX - read from offset %" PRIu32 " into buffer\n", offset);

            /* wait for event if the call is accepted */
            if (ucp_status.error == ERR_NONE)
            {
                while (event_callback == CLEAR_EVENT)
                {
                    __WFI();
                }
            }

            /* check status and actual read size */
            if ((event_callback == ARM_UC_PAAL_EVENT_READ_DONE) &&
                (buffer.size > 0))
            {
                /* the last page, in the last buffer might not be completely
                   filled, round up the program size to include the last page
                */
                uint32_t programOffset = 0;
                uint32_t programSize = (buffer.size + chunkSize - 1)
                                       / chunkSize * chunkSize;

                /* write one page at a time */
                while ((programOffset < programSize) &&
                       (retval == 0))
                {
                    retval = flash.program(&(buffer.ptr[programOffset]),
                                           app_start_addr + offset + programOffset,
                                           chunkSize);
#if 1
/** Debug code for illustrating flash failure.  Would not actually
   * call flash read here
   */
{
    int32_t status = 0;
    uint8_t confirm_buffer[chunkSize];
    uint32_t confirm_addr = app_start_addr + offset + programOffset;

    status = flash.read(confirm_buffer, 
                        confirm_addr,
                        chunkSize);

    if (memcmp(&(buffer.ptr[programOffset]), confirm_buffer, chunkSize)) {
        printf("XXX - error programming flash at %" PRIX32 "\n", 
              confirm_addr);
        printf("XXX - wrote    ["); for(uint32_t i=0; i<chunkSize; printf("%02x ", buffer.ptr[programOffset + i++])); printf("]\n");
        printf("XXX - but read ["); for(uint32_t i=0; i<chunkSize; printf("%02x ", confirm_buffer[i++])); printf("]\n");

        printf("XXX - this was while writng a chunk in the range %" PRIX32 " to %" PRIX32 "\n",
               app_start_addr + offset, app_start_addr + offset + programSize);
    }
    
}

#endif

                    programOffset += chunkSize;

#if defined(SHOW_PROGRESS_BAR) && SHOW_PROGRESS_BAR == 1
                    printProgress(offset + programOffset, details->size);
                    fota::printProgress(offset + programOffset,
                                        details->size,
                                        "Write FW");
#endif
                }

                tr_debug("\r\n%" PRIu32 "/%" PRIu32 " writing %" PRIu32 " bytes to 0x%08" PRIX32,
                         offset, (uint32_t) details->size, programSize, app_start_addr + offset);
                printf("\r\nXXX - %" PRIu32 "/%" PRIu32 " writing %" PRIu32 " bytes to 0x%08" PRIX32 "\n",
                         offset, (uint32_t) details->size, programSize, app_start_addr + offset);

                offset += programSize;
                printf("XXX - written %" PRIu32 " total bytes so far\n", offset);
            }
            else
            {
                tr_error("ARM_UCP_Read returned 0 bytes");

                /* set error and break out of loop */
                retval = -1;
                break;
            }
        }

        fota::printProgress(100, 100, "Write FW");

        result = (retval == 0);
    }

    return result;
}

```
The following is the log produced from the above code
[fota_1102i.log](https://github.com/ARMmbed/mbed-os/files/1447031/fota_1102i.log)

**Steps to reproduce**

Follow steps for FOTA in the readme for fota-demo application. (Although, as the above snippet indicates it should be easy to write a simple test program that writes a large amount of data and compares what was written to what was subsequently read.)

----------------------------------------------------------------

0xc0170, 2017-11-07:
@andreaslarssonublox @andreaspeterssonublox Can you please review this issue?

> Some debug has been done on the issue (details below).
Down-revving the flash driver per
ARMmbed/mbed-bootloader-internal#41
seems to eliminate the issue.

Please add code snippet here or more details for this one, private repository is not accessible for everyone and should not be referenced just via a link.

andreaslarssonublox, 2017-11-08:
@0xc0170, I guess u-blox don't need to look into this since there is a fix in #5455?

kylestein-arm, 2017-11-09:
I have cherry-piked the changes from pull-request 5455 and verified that it fixes my issue.
Thanks,

closed 2017-11-09 by kylestein-arm