ARMmbed/mbed-os #5441: Filesystem Class : Seek/Rewind

deepikabhavnani, 2017-11-06:
dir_seek / dir_rewind functions at present do not return status. Mechanism should be provided to return error status.

ChanFS does not support directory seek/rewind functionality, and we cannot return -ENOSYS to indicate function not implemented.

Second best option is to assert in filesystem.

@geky @sg- Please suggest

sg-, 2017-11-06:
why can't we return -ENOSYS ?

deepikabhavnani, 2017-11-06:
API return type is void https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/FileSystem.h#L241

sg-, 2017-11-07:
Sounds like we know what to do then.

geky, 2017-11-07:
Internally tracking [IOTMORF-1677](https://jira.arm.com/browse/IOTMORF-1677)

Original definitions were taken from posix: http://pubs.opengroup.org/onlinepubs/7908799/xsh/rewinddir.html

Since then we've learned that not having the ability to return an error was a mistake. Returning error codes on all functions greatly simplifies things.

Unfortunately changing it now will break inheritance compatibility.

One option is to set a flag so that readdir returns an error on the next call.

I think just changing these definitions to return error codes would be better long run, but we should probably wait for the next release since filesystem things are currently moving around.

closed 2017-11-21 by deepikabhavnani