ARMmbed/mbed-os #5443: Error when creating a second `AnalogIn` object

Ivan-Perez, 2017-11-07:
### Description
- Type: Bug
- Related issue: error introduced in commit https://github.com/ARMmbed/mbed-os/commit/8540e8a1dd7945791e0ab59ddbb46601e8f16ebd
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
STM32L0 (maybe other targets are affected too, I haven't checked it)

**Expected behavior**

When a second `AnalogIn` object is created (and it occurs for every object of that class created thereafter), its `handle` member is not being initialized in the constructor. The object starts using uninitialized memory and it works wrong.

The constructor calls the function `void analogin_init(analogin_t *obj, PinName pin)`, which relies on a global variable (`adc_inited`) to check if the `handle` member is initialized or not. The first `AnalogIn` object created works properly, because that variable is 0, but after that, no other new objects will initialize that member. So this is wrong.

Overwriting the value of `adc_inited` to `0` before creating a new `AnalogIn` object fixes the issue, but I think there should be a better fix to avoid the problem.

Could this be fixed?

0xc0170, 2017-11-07:
There was a similar report, that was reported right after this one. Your assumptions look fine, I had a quick look at the analog in implementation for L0.

@bcostm @adustm @LMESTM @jeromecoutant please review

Ivan-Perez, 2017-11-07:
Yeah, it was a teammate who realized about the same issue and posted it here too. I was a little faster than him :)

bcostm, 2017-11-07:
I don't remember why we should init the ADC only once for L0, L1, L4, F0, F1 ? It is not done for F2, F4, F7.

```
    // The ADC initialization is done once
    if (adc_inited == 0) {
        adc_inited = 1;
```

I think we can remove these lines and the adc_inited variable.

bcostm, 2017-11-07:
Can you please check my branch and see if it solves your issue ?

[https://github.com/bcostm/mbed-os/tree/fix_adc_inited]()

If it is ok, I will do the same for all other STM32 series.

Ivan-Perez, 2017-11-07:
We've tried it but it isn't working. The member `lock` (and possibly the other members in the `ADC_HandleTypeDef` struct) isn't initialized. The only one being initialized right now is `State`.

Moving the following piece of code out of the condition `if (adc_inited == 0)` seems to fix the issue for us.

```cpp
        if (HAL_ADC_Init(&obj->handle) != HAL_OK) {
            error("Cannot initialize ADC");
        }
```

Another possibility is to add `&obj->handle->Lock = HAL_UNLOCKED;` before `if (adc_inited == 0)` condition. This would allow to read from analog input (as it is unlocked), but there are other uninitialized members, so it may not work always as expected.

bcostm, 2017-11-08:
The problem is that I cannot make my test program fail. In all cases the ADC instances are correctly initialized, I tried with the same GCC version as you use without success. Can you please share your test program ?

bcostm, 2017-12-04:
Can you please close this issue as it has been solved. Thanks.

ST_TO_BE_CLOSED

closed 2017-12-04 by 0xc0170