ARMmbed/mbed-os #5444: Bug library AnalogIn

VictorEnc, 2017-11-07:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Related issue: 
- Priority: Major

---------------------------------------------------------------
## Bug

When initially creating an AnalogIn object the global variable adc_inited is equal to 0. and that causes the handle of analogin_t to initialize correctly. when destroying the previous object and performing operations the memory area where the handle was modified, when creating another AnalogIn object a random memory address is assigned to the handle that may or may not coincide with the previous address and since the global adc_inited variable is equal to 1, the handle is not initialized correctly and in some cases proving that the analog can not be read correctly.

**Target**
STM32L073

**Compiler **
gcc-v5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]

0xc0170, 2017-11-07:
> gcc-v5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]

Please use gcc v6 as that one is supported version.

I had a quick look for L0 analog in implementation and looks like there is a bug. As object memebers are initialized but only the very first time, then another object would not get this initialization (this probably should be per ADC module, and kept somewhere in memory or initialized/deinitialized as per ADC module pins are init/free).

@bcostm @adustm @LMESTM @jeromecoutant

0xc0170, 2017-11-07:
This looks like a duplicate of https://github.com/ARMmbed/mbed-os/issues/5444 ,please verify and we can close this and continue in the earlier issue.

closed 2017-11-07 by VictorEnc