ARMmbed/mbed-os #5448: Add AnalogIn interrupt to save power

adustm, 2017-11-07:
### Description
- Type: Enhancement
- Related issue: https://os.mbed.com/questions/79353/How-to-use-the-analog-comparators-of-thi
- Priority: Major

## Enhancement

**Reason to enhance or problem with existing solution**
Mbed-os does not offer an interrupt for analogin objects. This is a problem for battery powered MCUs that could wake-up only when needed instead of pooling the ADC with a timer interrupt.

**Suggested enhancement**
Support comparator interrupt would allow the wakeup of the MCU only when needed
**Pros**
Save power, which is important in the context of IOT devices.

**Cons**

0xc0170, 2017-11-07:
Thanks for the proposing the interrupt API for `analogin`.

cc @c1728p9 @bulislaw

jeppefrandsen, 2018-11-07:
@adbridge do you already have a plan for supporting this? Thanks a lot

adbridge, 2018-11-07:
@jeppefrandsen It has been captured as a requirement but we have no current details for when it may come in.

closed 2019-10-31 by ciarmcom