ARMmbed/mbed-os #5450: NUCLEO_F429ZI: Flash memory bank 2 sector sizes incorrect

paviitan, 2017-11-07:
### Description

NUCLEO_F429ZI Flash sector size is incorrect for Bank 2, Sectors 12-15. This causes problems when using FlashIAP to read/write on MCU Flash. See table 6 on page 77 [RM0090 PDF](http://www.st.com/content/ccc/resource/technical/document/reference_manual/3d/6d/5a/66/b4/99/40/d4/DM00031020.pdf/files/DM00031020.pdf/jcr:content/translations/en.DM00031020.pdf)

Bug exists in release 5.6.3

- Type: Bug 

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_F429ZI

**Toolchain:**
GCC_ARM 

**mbed-os (5.6.3) sha:**
e62a1b923
**Suspected cause** 
`GetSector` function in `/mbed-os/targets/TARGET_STM/TARGET_STM32F4/flash_api.c` returns “16” for sectors 12 - 15

**Steps to reproduce:**
Run simple code to check sectors by start address:
``` C++
#include "mbed.h"

FlashIAP flash;

int main(void){
    uint32_t first_sector = (uint32_t)0x08000000;
    uint32_t last_sector = (uint32_t)0x081E0000;
    uint32_t addr = first_sector;
    uint32_t sector_size;
    printf("\r\nStart of program\r\n");
    while(addr < last_sector +1){
        sector_size = flash.get_sector_size(addr);
        printf("Sector at %#10x of size %u\r\n", addr, sector_size);
        addr += sector_size; // next sector
    }
    printf("\r\nEnd of program\r\n");
    return 0;
}

``` 
**Actual result:**
``` OUTPUT

Start of program
Sector at  0x8000000 of size 16384
Sector at  0x8004000 of size 16384
Sector at  0x8008000 of size 16384
Sector at  0x800c000 of size 16384
Sector at  0x8010000 of size 65536
Sector at  0x8020000 of size 131072
Sector at  0x8040000 of size 131072
Sector at  0x8060000 of size 131072
Sector at  0x8080000 of size 131072
Sector at  0x80a0000 of size 131072
Sector at  0x80c0000 of size 131072
Sector at  0x80e0000 of size 131072
Sector at  0x8100000 of size 65536 // Erroneus sector / size
Sector at  0x8110000 of size 65536 // Sector 16
Sector at  0x8120000 of size 131072
Sector at  0x8140000 of size 131072
Sector at  0x8160000 of size 131072
Sector at  0x8180000 of size 131072
Sector at  0x81a0000 of size 131072
Sector at  0x81c0000 of size 131072
Sector at  0x81e0000 of size 131072

End of program
``` 
 
**Expected result:**
``` OUTPUT
Start of program
Sector at  0x8000000 of size 16384
Sector at  0x8004000 of size 16384
Sector at  0x8008000 of size 16384
Sector at  0x800c000 of size 16384
Sector at  0x8010000 of size 65536
Sector at  0x8020000 of size 131072
Sector at  0x8040000 of size 131072
Sector at  0x8060000 of size 131072
Sector at  0x8080000 of size 131072
Sector at  0x80a0000 of size 131072
Sector at  0x80c0000 of size 131072
Sector at  0x80e0000 of size 131072
Sector at  0x8100000 of size 16384 // Sector 12
Sector at  0x8104000 of size 16384
Sector at  0x8108000 of size 16384
Sector at  0x810c000 of size 16384
Sector at  0x8110000 of size 65536 // Sector 16
Sector at  0x8120000 of size 131072 
Sector at  0x8140000 of size 131072
Sector at  0x8160000 of size 131072
Sector at  0x8180000 of size 131072
Sector at  0x81a0000 of size 131072
Sector at  0x81c0000 of size 131072
Sector at  0x81e0000 of size 131072

End of program
```

LMESTM, 2017-11-07:
@paviitan  thanks for reporting the issue.
There might indeed to be an error - not sure why this isn't seen with flash tests though. 
By reading the code I would say that the fix could be as shared below, but I will need time in the coming days to verify and test. In case this is blocking for you, you may want to try;
$ git diff TARGET_STM32F4/flash_api.c
diff --git a/targets/TARGET_STM/TARGET_STM32F4/flash_api.c b/targets/TARGET_STM/TARGET_STM32F4/flash_api.c
index 1455f07..aea82d8 100644
--- a/targets/TARGET_STM/TARGET_STM32F4/flash_api.c
+++ b/targets/TARGET_STM/TARGET_STM32F4/flash_api.c
@@ -181,7 +181,7 @@ static uint32_t GetSector(uint32_t address)
 #if defined(ADDR_FLASH_SECTOR_16)
     if (address & 0x100000) { // handle 2nd bank
         sector = FLASH_SECTOR_12;
-        tmp = address - ADDR_FLASH_SECTOR_12;
+        tmp = address - 0x100000;
     }
 #endif
     if (address < ADDR_FLASH_SECTOR_4) { // 16k sectorsize

0xc0170, 2017-11-07:
Is this bug related to this one https://github.com/ARMmbed/mbed-os/issues/5439 (same family, having problems  with flash algo, regarding the second bank as the report specifies) ?

paviitan, 2017-11-08:
@0xc0170 Hi,
I would say the issues are related to #5439 

- Ublox Odin W2 seems to be using the same ST Firmware library

- STM32F439 has same 2Mbyte flash module

LMESTM, 2017-11-08:
@paviitan  @0xc0170 
PR #5455 available for early tests on your side. 
Non-regression automated tests started meanwhile.

closed 2017-11-10 by 0xc0170