ARMmbed/mbed-os #5451: ATCmdParser::vrecv() won't work

betzw, 2017-11-07:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
all

**Toolchain:**
all

**Actual behavior**
Function `ATCmdParser::vrecv()` does not work with responses like e.g. `"OK%*[\r]"` as carriage returns (`CR`) when read from serial will be transformed into line feeds (`LF` - aka. `'\n'`, see [here](https://github.com/ARMmbed/mbed-os/blob/master/platform/ATCmdParser.cpp#L256-L265)). 
As a consequence `scanf()` will never match all characters in the response (see [here](https://github.com/ARMmbed/mbed-os/blob/master/platform/ATCmdParser.cpp#L301-L305)).

0xc0170, 2017-11-08:
@hasnainvirk Please review

hasnainvirk, 2017-11-09:
There is some effort going on to fix issues in ATCmdParser. @kjbracey-arm,   Teemu Kultala and Teppo, are the guys responsible now. I can't find them on git. Will update them later

kjbracey-arm, 2017-11-09:
Due to the conversion you should be using `\n` in the match string.

kjbracey-arm, 2017-11-09:
https://stackoverflow.com/questions/21057780/why-scanf-skips-r-but-reads-n

betzw, 2017-11-09:
Well, `sscanf()` treats all (non-escaped) whitespace characters any amount of (any kind of) white space, including none ...
Anyway, your workaround proposal should work.

closed 2017-11-09 by betzw

kjbracey-arm, 2017-11-09:
I don't recall the full detail, but there was another change between `ATParser` and `ATCmdParser` - it now has two different modes of operation - matching prefix and matching whole line. The original only matched prefixes. I believe some of the funny `%*[` constructs in `ATParser` users were trying to get it to consume a line properly.

Now if you just pass a string containing a straight `\n`, it will require a full-line match. This then means `recv("Foo: %s\n")` will work, which it didn't in `ATParser`.