ARMmbed/mbed-os #5461: linker script, stm32f412xx.icf, is wrong

sus-e, 2017-11-09:
I think I found some minor problem on your IAR's linker file for stm32f412xG, github path is
mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F412xG/device/TOOLCHAIN_IAR/stm32f412xx.icf

This chip has 256kbyte RAM and comment of linker file says so. But layout is for 128Kbyte RAM,
below is current definition
/* [RAM = 256kb = 0x40000] Vector table dynamic copy: 113 vectors = 452 bytes (0x1C4) to be reserved in RAM */
define symbol __NVIC_start__ = 0x20000000;
define symbol __NVIC_end__ = 0x200001C7; /* Aligned on 8 bytes */
define symbol __region_RAM_start__ = 0x200001C8;
define symbol __region_RAM_end__ = 0x2001FFFF;

but I think it must be below
/* [RAM = 256kb = 0x40000] Vector table dynamic copy: 113 vectors = 452 bytes (0x1C4) to be reserved in RAM */
define symbol __NVIC_start__ = 0x20000000;
define symbol __NVIC_end__ = 0x200001C7; /* Aligned on 8 bytes */
define symbol __region_RAM_start__ = 0x200001C8;
define symbol __region_RAM_end__ = 0x2003FFFF; /* <<<<<< for 256Kbyte */

I hope this report helps your team.

jeromecoutant, 2017-11-09:
Thx for the issue report

Please check #5463

bcostm, 2017-11-14:
This issue has been solved. Can you please close it now ? Thanks.
ST_TO_BE_CLOSED

closed 2017-11-14 by 0xc0170

sus-e, 2017-11-15:
thanks for fix it.