ARMmbed/mbed-os #5462: initial pin value not retained

thinkberg, 2017-11-09:
### Description
- Type: Bug 
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
NRF52

**mbed-os sha:**
`e62a1b923 (HEAD -> mbed-os-5.6, tag: mbed_lib_rev154, tag: mbed-os-5.6.3)`

**Expected behavior**

`DigitalOut led(LED1, 1);` should have the pint be set to HIGH output.

**Actual behavior**

In the `gpio_mode()`, the pin is resetted (`gpiote_pin_uninit`) and the output value gets lost during initialization of `DigitalOut` as it is not saved anywhere in the config.

**Steps to reproduce**

Simply create a DigitalOut led(pin, 1) with an LED connected to the pin that requires HIGH to be off.

[image]

0xc0170, 2017-11-09:
@thinkberg Just to confirm, this looks like it is NRF5x target problem in the HAL rather than generic. From the look at the code, `gpiote_pin_uninit` is in the HAL.

@anangl @nvlsianpu

nvlsianpu, 2017-11-13:
@thinkberg https://github.com/ARMmbed/mbed-os/pull/5487

nvlsianpu, 2017-11-17:
@thinkberg - Is fixed, can you close the issue.

closed 2017-11-17 by thinkberg