ARMmbed/mbed-os #5465: us_ticker single test failing in master branch

akrnvm, 2017-11-09:
### Description
- Type: Bug 
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
Custom Board

**Toolchain:**
GCC_ARM|ARM|IAR

**Toolchain version:**
5.21
**mbed-cli version:** 
1.2.2

**Expected behavior**
ticker attach and wait function should execute propperly 

**Actual behavior**
ticker interrupt is getting corrupted by wait function

**Steps to reproduce**
python tools/singletest.py --auto -v --tc ARM -n MBED_11

```
HOST: Reset target...
HOST: Property 'timeout' = '15'
HOST: Property 'host_test_name' = 'wait_us_auto'
HOST: Property 'description' = 'Ticker Int'
HOST: Property 'test_id' = 'MBED_11'
HOST: Start test...
* in 0.00 sec (1.00) [FAIL]
* in 0.00 sec (1.00) [FAIL]
* in 0.00 sec (1.00) [FAIL]
* in 0.00 sec (1.00) [FAIL]
* in 0.00 sec (1.00) [FAIL]
* in 0.00 sec (1.00) [FAIL]
* in 0.00 sec (1.00) [FAIL]
* in 0.00 sec (1.00) [FAIL]
* in 0.00 sec (1.00) [FAIL]
* in 0.00 sec (1.00) [FAIL]
* in 0.00 sec (1.00) [FAIL]
* in 0.00 sec (1.00) [FAIL]
* in 0.00 sec (1.00) [FAIL]
Consecutive OK timer reads: 0
Completed in 0.01 sec

{{failure}}
{{end}}
Test::Output::Finish
```

Sample code to test 
```
#include "mbed.h"

Ticker flipper_1;
Ticker flipper_2;
DigitalOut led1(LED1);
DigitalOut led2(LED2);

void flip_1() {
        led1 = !led1;
}

void flip_2() {
         led2 = !led2;
}

int main() {
    led1 = 0;
    led2 = 0;
    flipper_1.attach(&flip_1, 1.0); 
    flipper_2.attach(&flip_2, 2.0); 

    while (true) {
      wait(1.0);
    }
}
```
Notes:
1. test case is passed  without wait() function.

2. **test case is passing with mbed 5.6.3 release package and failing in mbed master branch.**

-----------------------------------------------------------------

## Question

How to fix this issue?

0xc0170, 2017-11-09:
> Custom Board

What is the target (is that HAL implementation is not in this codebase) ?

akrnvm, 2017-11-09:
I have using customised board of cortex-M4, all other timer related test cases and greentea test cases are passing.
Only one failure in Single test cases(MBED_11) without RTOS.
Same testcase is Passing  with RTOS. 
Testcase is PASSING in mbed 5.6.3 with and without RTOS.
Is any changes in usticker  from mbed 5.6.3 release package to master branch

cmonr, 2017-12-07:
@akrnvm Could you retry the test on mbed-os-5.6.6 and confirm this is still an issue? Have there also been any developments on your end?

akrnvm, 2017-12-08:
@cmonr  I have tested with mbed-os-5.6.6 and the test case is passing with Keil and gcc_arm.
Note:
Test case is failing in IAR.
Test case is failing in latest master branch with all toolchains (ARM,GCC,IAR)

akrnvm, 2018-01-22:
ISSUE #5842 

It related to above issue(5842) and after the fix ( removing wait() function )test cases are passing.
Closing this ticket.

closed 2018-01-22 by akrnvm