ARMmbed/mbed-os #5467: mbed export not working with uvision5

rspelta, 2017-11-09:
### Description
- Type: Bug
- Targets "tested": NUCLEO_L476ZI, SILICA_SENSOR_NODE
- IDE used: uVision 5.24
- Project: mbed-os-example-blinky (lastest version)

Marcelo has asked to me to open this issue.
cc: @MarceloSalazar 

**Issue:**
When I export the project [(mbed-os-example-blinky)](https://github.com/ARMmbed/mbed-os-example-blinky) for uVision5 and I try to debug it, I get this error:

[image]

it’s something to do with the flash algorithm not being set correctly. It seems that the project is exported wrongly. 

**My temporary solution:**
I find out a workaround setting manually the project. Here my steps:
1. mbed import https://github.com/ARMmbed/mbed-os-example-blinky 
2. mbed export -i uvision5 -m SILICA_SENSOR_NODE
3. open the project and if needed download the libreries required
4. Even if you have this error, ignore it
[image]
5. go to "Project -> Options for the target 'mbed-os-example-blinky'"
6. Click on "Debug" tab
[image]
7. Click on "Settings" button, it is near "Use: ST-Link Debugger"
8. Select "Connect: with pre-reset" (at the bottom of the form, in "Connect & Reset Options")
[image]
9. Click on "Flash download" tab
10. Click on "add" button
[image]
11. Select "STML4xx 1MB Flash", it is the algorithm. Now it is setted correctly!
[image]

**Side note:**  When I open the project in uvision, it complains about a missing device pack for STM32L4 even though it HAS been installed. Even if you try and re-install the pack completely, it still complains every time you open the project. About this issue seems you already know this problem and it will be fixed in the next release.

theotherjimmy, 2017-11-09:
Thanks for the detailed description of the workaround! I'll take a look at this in the coming days. I may not have time this week, but you should certainly hear from me sometime next week.

theotherjimmy, 2017-11-13:
@rspelta @MarceloSalazar I was able to track down part of this issue. Could you take a look at #5489 for me?

cmonr, 2018-07-09:
@MarceloSalazar Fyi, we removed the IOTMORF-2318 comment to make sure the issue script doesn't do weird things.

VVESTM, 2019-03-19:
Hi, 
For information, the problem is still present. I try to work on Disco F746 and I am not able to flash the SW with Keil. (The proposed workaround is working fine).
By the way, I have also problems to export to IAR... Do you have any tests on export feature ? 
Regards.

theotherjimmy, 2019-03-19:
@VVESTM How did you test it? The online exporters have not been updated yet. We test IAR _building_.

VVESTM, 2019-03-19:
@theotherjimmy , I clone mbed-os on my laptop, then, I use this command : 
    mbed export --ide uvision6 --target DISCO_F746NG
or --ide IAR

theotherjimmy, 2019-03-19:
@VVESTM The exporters don't support exporting just Mbed OS. Be sure to include a project with a main.cpp or main.c as well.

VVESTM, 2019-03-19:
@theotherjimmy, forgot to mention this, sorry. I also added a main.cpp in mbed-os root directory. (only an hello world...)

theotherjimmy, 2019-03-19:
Thanks for clarifying. Are you using master branch?

VVESTM, 2019-03-19:
yes, I am on master branch. We can continue by private mail if needed in order not to pollute this thread.

theotherjimmy, 2019-03-19:
@VVESTM This is about as on topic as it gets. Let's keep this discussion public. What's the error message that you're seeing?

VVESTM, 2019-03-19:
OK, so, I am on master branch and I added a main.cpp file. I have also mbed_setting.py to have toolchain paths.
I export the project to Keil with this command : 
    mbed export --ide uvision6 --target DISCO_F746NG
Then, I open the .uvprojx project with Keil. Compilation is OK.
When trying to debug the source code, I have a popup with this message : 
     Error: Flash Download failed - "Cortex-M7"

If I apply @rspelta workarround, I am able to debug.

theotherjimmy, 2019-03-19:
Do you get the error message in step 4?

VVESTM, 2019-03-20:
No, I can open the project file and compile it without error. Problem occurs when I want to start the debugger (or flash the board)

bridadan, 2019-03-25:
I believe this comes from an issue in the current device index. The current device index does not mark any flash algorithm as "default", so no flash algorithm is set in the exported projects.

I believe the following changes would need to happen to fix this:

- Need to update Mbed OS to use version 0.2.4 of CPM: https://github.com/ARMmbed/cmsis-pack-manager/releases/tag/v0.2.4
- Need to regenerate device index with this version of CPM

PR #10409 (2019-04-18): Fixing default algorithms in cmsis pack index
### Description

Fixes #5467.

For certain targets, they did not have any "default" flash algorithms, which meant the target exported without an flash algorithm selected. This updates the device index with the latest output from the pack manager.

This is again a massive diff, so I'm attaching the patch since github probably won't render it.
[diff.txt](https://github.com/ARMmbed/mbed-os/files/3082405/diff.txt)

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test upd [... cut]
closed 2019-04-18 by 0xc0170