ARMmbed/mbed-os #5477: tests-mbedmicro-rtos-mbed-timer tests failing on REALTEK_RTL8195AM

studavekar, 2017-11-10:
### Description
tests-mbedmicro-rtos-mbed-timer fails intermittently , This will gate us addition of REALTEK_RTL8195AM to CI

To reproduce run below command in  loop 
```
mbed test -m REALTEK_RTL8195AM -n tests-mbedmicro-rtos-mbed-timer -t GCC_ARM -v

```

- Type: Bug
- Priority:  Major 
---------------------------------------------------------------
## Bug

**Target**
REALTEK_RTL8195AM

**mbed-cli version:** 
12.0

**mbed-os sha:**
47bae16

**DAPLink version:**
0244
```
+---------------------------+-------------------+---------------------------------+----------------------------------------------------------------+--------+--------+---------+--------------------+
| target                    | platform_name     | test suite                      | test case                                                      | passed | failed | result  | elapsed_time (sec) |
+---------------------------+-------------------+---------------------------------+----------------------------------------------------------------+--------+--------+---------+--------------------+
| REALTEK_RTL8195AM-GCC_ARM | REALTEK_RTL8195AM | tests-mbedmicro-rtos-mbed-timer | Test: Timer (based on os ticker) - float operator.             | 1      | 0      | OK      | 0.04               |
| REALTEK_RTL8195AM-GCC_ARM | REALTEK_RTL8195AM | tests-mbedmicro-rtos-mbed-timer | Test: Timer (based on os ticker) - measured time accumulation. | 1      | 0      | OK      | 1.18               |
| REALTEK_RTL8195AM-GCC_ARM | REALTEK_RTL8195AM | tests-mbedmicro-rtos-mbed-timer | Test: Timer (based on os ticker) - reset.                      | 1      | 0      | OK      | 0.04               |
| REALTEK_RTL8195AM-GCC_ARM | REALTEK_RTL8195AM | tests-mbedmicro-rtos-mbed-timer | Test: Timer (based on os ticker) - start started timer.        | 1      | 0      | OK      | 0.05               |
| REALTEK_RTL8195AM-GCC_ARM | REALTEK_RTL8195AM | tests-mbedmicro-rtos-mbed-timer | Test: Timer (based on os ticker) is stopped after creation.    | 1      | 0      | OK      | 0.04               |
| REALTEK_RTL8195AM-GCC_ARM | REALTEK_RTL8195AM | tests-mbedmicro-rtos-mbed-timer | Test: Timer (based on user ticker) - float operator.           | 1      | 0      | OK      | 0.02               |
| REALTEK_RTL8195AM-GCC_ARM | REALTEK_RTL8195AM | tests-mbedmicro-rtos-mbed-timer | Test: Timer (based on user ticker) - reset.                    | 1      | 0      | OK      | 0.02               |
| REALTEK_RTL8195AM-GCC_ARM | REALTEK_RTL8195AM | tests-mbedmicro-rtos-mbed-timer | Test: Timer (based on user ticker) - start started timer.      | 1      | 0      | OK      | 0.02               |
| REALTEK_RTL8195AM-GCC_ARM | REALTEK_RTL8195AM | tests-mbedmicro-rtos-mbed-timer | Test: Timer (based on user ticker) is stopped after creation.  | 1      | 0      | OK      | 0.02               |
| REALTEK_RTL8195AM-GCC_ARM | REALTEK_RTL8195AM | tests-mbedmicro-rtos-mbed-timer | Test: Timer (based on user ticker) measured time accumulation. | 1      | 0      | OK      | 0.02               |
| REALTEK_RTL8195AM-GCC_ARM | REALTEK_RTL8195AM | tests-mbedmicro-rtos-mbed-timer | Test: Timer - time measurement 1 ms.                           | 1      | 0      | OK      | 0.02               |
| REALTEK_RTL8195AM-GCC_ARM | REALTEK_RTL8195AM | tests-mbedmicro-rtos-mbed-timer | Test: Timer - time measurement 1 s.                            | 0      | 0      | SKIPPED | 0.0                |
| REALTEK_RTL8195AM-GCC_ARM | REALTEK_RTL8195AM | tests-mbedmicro-rtos-mbed-timer | Test: Timer - time measurement 10 ms.                          | 0      | 1      | FAIL    | 0.04               |
| REALTEK_RTL8195AM-GCC_ARM | REALTEK_RTL8195AM | tests-mbedmicro-rtos-mbed-timer | Test: Timer - time measurement 100 ms.                         | 0      | 0      | SKIPPED | 0.0                |
+---------------------------+-------------------+---------------------------------+----------------------------------------------------------------+--------+--------+---------+------------------
```

Logs : [REALTEK_RTL8195AM.log](https://github.com/ARMmbed/mbed-os/files/1462906/REALTEK_RTL8195AM.log)

Archcady, 2017-12-14:
Are these tests still failing with the latest version of MBED-OS

studavekar, 2017-12-20:
@Archcady since last time test structure has been changed, latest run on master i don't see a failure 

[REALTEK_RTL8195AM-tests-mbed_drivers-timer.log](https://github.com/ARMmbed/mbed-os/files/1576765/REALTEK_RTL8195AM-tests-mbed_drivers-timer.log)

studavekar, 2017-12-20:
closing this issue as its not seen on master.

closed 2017-12-20 by studavekar