ARMmbed/mbed-os #5478: Add support for Waveshare BLE400

jeanmatthieud, 2017-11-11:
### Description
- Type: Enhancement

## Enhancement
Is it possible to add support for [Waveshare BLE400](https://www.waveshare.com/wiki/BLE400) board ?
I tried to do it myself but without success.

The Pins should be edited in this way:
```
    P00  = p0,
    P01  = p1,
    P02  = p2,
    P03  = p3,
    P04  = p4,
    P05  = p5,
    P06  = p6,
    P07  = p7,

    P08  = p8,
    P09  = p9,
    P010 = p10,
    P11 = p11,
    P12 = p12,
    P13 = p13,
    P14 = p14,
    P15 = p15,

    P16 = p16,
    P17 = p17,
    P18 = p18,
    P19 = p19,
    P20 = p20,
    P21 = p21,
    P22 = p22,
    P23 = p23,

    P24 = p24,
    P25 = p25,
    P26 = p26,
    P27 = p27,
    P28 = p28,
    P29 = p29,
    P30 = p30,

    LED0    = p18,
    LED1    = p19,
    LED2    = p20,
    LED3    = p21,
    LED4    = p22,

    BUTTON1 = p16,
    BUTTON2 = p17,

    KEY1 = p16,
    KEY2 = p17,
```

I'm not sure for RX, TX and CTS, RDS, but if the changes are made, I could test it and correct the files.

0xc0170, 2017-11-16:
Have you asked Waveshare if they are considering this addition ?

> I tried to do it myself but without success.

You can always share your current work done (in case there is more work needed besides the pin names you shared above). Is there?

> I'm not sure for RX, TX and CTS, RDS, but if the changes are made, I could test it and correct the files.

It seems there is Uart to usb converter on board, but not interface chip. No drag-n-drop programming?

scttnlsn, 2018-03-12:
I've also tried running mBed on this board and I'm not sure why it doesn't work properly.

```c
#include <mbed.h>

DigitalOut pin(P0_19);

int main() {
  while (true) {
    pin = 1;
    wait(0.5);
    pin = 0;
    wait(0.5);
  }

  return 0;
}
```

I flashed the above code with an ST-Link v2 but it did not work.  Haven't had a chance to debug it yet.

danielkucera, 2018-03-12:
You should try target  BBC micro:bit. It worked for me.

scttnlsn, 2018-03-12:
@danielkucera That worked for me as well.  Thanks!  Maybe it was just an issue w/ the board's configured clock speed?

danielkucera, 2018-03-12:
@scttnlsn No problem, I spent hours on making it work so I'm glad it helped. I don't know where the problem is, you can compare configs for those targets to see.
Here's my comment on mbeds page btw: https://os.mbed.com/forum/platform-39-Nordic-nRF51822-community/topic/4982/?page=2#comment-55087

ashok-rao, 2018-08-02:
There are currently no plans to support this target in Mbed OS. Issue can be closed. Please reopen if required. Thanks.

closed 2018-08-02 by 0xc0170

SoniaZotz, 2018-08-06:
So how does one develop embedded systems - where the board configuration does not follow any of Mbed standard defines boards?

ashok-rao, 2018-08-06:
@SoniaZotz : You are definitely welcome to port Mbed OS to this new target (& we'd be happy to answer questions you may have in the porting process), however, the above comment was to mention that there are no plans to add this target to the list of officially supported platforms from Mbed side..

0xc0170, 2018-08-06:
@SoniaZotz If you have any questions , create a new issue with specific question regarding porting new target. Good starting point is https://os.mbed.com/docs/latest/reference/porting-targets.html

danielkucera, 2018-08-06:
@SoniaZotz, I'm willing to help with porting and/or testing. Maybe we could start by forking `BBC micro:bit` which works for me and then just edit description, pin mapping etc.

SoniaZotz, 2018-08-07:
thank you all for your quick response. It answered a few questions for me and resulted in a few more.
1- who controls officially supported platforms from Mbed side
2- the porting process is rather complicated ie  https://os.mbed.com/docs/latest/reference/porting-targets.html
3- how does one port a module - for a specific processor ie: the NRF52 series as done in the NINA-
B1 series

ashok-rao, 2018-08-08:
@SoniaZotz .. 
1. Mbed internal teams controls the decision on officially supported platforms based on various criteria - https://www.mbed.com/en/about-mbed/mbed-enabled/ 
2. The porting process linked above is for a new MCU. You have to start from scratch if you are porting a new MCU to Mbed OS.
3. For a module (i.e. a derived target of an existing MCU port), we are working on a porting guide. This will be available ASAP. Hope this helps?

cc : @AnotherButler & @bulislaw (for point 3, refer: https://github.com/ARMmbed/mbed-os-5-docs/pull/625)

0xc0170, 2018-08-08:
> 2- the porting process is rather complicated ie https://os.mbed.com/docs/latest/reference/porting-targets.html

Please provide feedback to https://github.com/ARMmbed/mbed-os-5-docs about the documentation (create new issue if its already not there with suggestions or questions you have regarding this porting doc page).

mutantcybertechnologies, 2018-09-06:
If the BBC microbit setting works for the ble400 (technically the core51822r3),  has anyone tried to see if the Switch Science ty51822r3 can also be used.  The specs on this seem a closed match than those of the BBC microbit (32k ram vs 16th ram), and this would allow mbed os 5 to be used