ARMmbed/mbed-os #5479: Build error occurs by using GR-PEACH board on Online Compiler

TomoYamanaka, 2017-11-13:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
GR-PEACH

Currently, compiling any mbed samples that older than Mbed OS version 5.4.7 by using GR-PEACH board on Mbed Online compiler, the below build error occurs.
`The build system did not finish successfully.`

The situation of this issue is the below.
- If memory serves, there was no problem until last month(Oct.).
- Changing to another target board(ex)  K64F, LPC1768, etc), the compile is successed.
- this error occurs no matter what Mbed OS version 5.4.7 or older (Mbed Classic is also the same).

sarahmarshy, 2017-11-15:
Looks like renesas has issued a warning - https://os.mbed.com/platforms/Renesas-GR-PEACH/#important-notice

TomoYamanaka, 2017-11-16:
GR-PEACH is unsupported temporarily in Mbed OS 5.5 or later. For this reason, previously, GR-PEACH could be compiled using Mbed OS 5.4 both on Online and on Offline. However, currently, even if using Mbed OS 5.4, GR-PEACH can not be compiled on Online. 

It seems that this problem blocks GR-PEACH users’ activity.
Thus, if you could update us on this soon, it would be very helpful.

theotherjimmy, 2017-11-16:
Hi @TomoYamanaka 

I'm looking into this today. Thanks for your patience.

TomoYamanaka, 2017-11-21:
How is going? Any update?

GR-PEACH users are in trouble that can not compile on Online environment.
So I will appreciate if you solve this issue promptly.

theotherjimmy, 2017-11-21:
Hi @TomoYamanaka,

I'm debugging the issue on partnerbeta right now. This has been very hard to track down.

th3james, 2017-11-21:
@TomoYamanaka @theotherjimmy The fix is now live on os.mbed.com

theotherjimmy, 2017-11-21:
Thanks @th3james! 🥇  that was quick!

TomoYamanaka, 2017-11-22:
@theotherjimmy @th3james 

Thank you for your prompt reply.
I confirmed to compile the programs by using GR-PEACH board on Mbed Online compiler.
I close this.

closed 2017-11-22 by TomoYamanaka