ARMmbed/mbed-os #5480: MAX32630 BLE - onDataSent callback not called

janjongboom, 2017-11-13:
### Description
- Type: Bug

---------------------------------------------------------------
## Bug

**Target**
MAX32630FTHR

**Toolchain:**
Unknown

**mbed-os sha:**
It says 'latest mbed', but not sure which one they mean.

According to https://os.mbed.com/questions/79492/blegattServeronDataSentonDataSent-never-/ `onDataSent` callback is no longer invoked on this MAXIM target.

0xc0170, 2017-11-13:
This is not a sufficient bug report that would allow us to reproduce this. We should know the toolchain version, have the versions for all the libs (in this case BLE and mbed-os) and also code snippet to be able to reproduce the issue (I can see they provided some code in the question, but was not copied here).

Note: we are working on providing a template that will define how a bug report should look like 📨

janjongboom, 2017-11-15:
@0xc0170 Asked for some more details in the original question. I can't verify it myself as I don't have the board. Figure I forward these to the bug tracker here regardless, so we can at least track it.

radekkierner, 2017-11-16:
mbed-os sha:
5.6.4 rev.78474a5

Below is my code exported from online compiler. I also noticed that write callback is only called when write  data length is equal of array size in characteristic declaration

[main.txt](https://github.com/ARMmbed/mbed-os/files/1478222/main.txt)

0xc0170, 2017-11-16:
> @0xc0170 Asked for some more details in the original question. I can't verify it myself as I don't have the board. Figure I forward these to the bug tracker here regardless, so we can at least track it.

Thanks! we just need to start having formalized tracking reports. Without sufficient details for reproducing this, it would become idle and never fixed. 

@radekkierner Thanks for the program

Attaching it here

```
#include "mbed.h"
#include "ble/BLE.h"

DigitalOut led(LED1, 1);
DigitalOut led2(LED2, 1);

const static char* SERVICE_UUID = "32372fb9-5f73-4c32-a17b-25e17f52a99a";
static uint8_t service_data[16];

const static char* CHARACTERISTIC_UUID = "1d6ba3db-d405-4034-96fc-78942ef7075f";

const static char DEVICE_NAME[] = "test";

static const uint8_t DATA_MTU = 20;

Serial pc(USBTX, USBRX);

UUID service_uuid =  UUID(SERVICE_UUID);
UUID characteristic_uuid = UUID(CHARACTERISTIC_UUID);

static uint8_t readValue[DATA_MTU] = {0};

Gap::ConnectionParams_t fast;

Ticker ticker;

//ReadOnlyArrayGattCharacteristic<uint8_t, sizeof(readValue)> readChar(characteristic_uuid, readValue, GattCharacteristic::BLE_GATT_CHAR_PROPERTIES_NOTIFY ,NULL,0);
//WriteOnlyArrayGattCharacteristic<uint8_t, sizeof(writeValue)> writeChar(characteristic_uuid, writeValue, GattCharacteristic::BLE_GATT_CHAR_PROPERTIES_WRITE_WITHOUT_RESPONSE ,NULL,0);

ReadWriteArrayGattCharacteristic<uint8_t, sizeof(readValue)> readChar(characteristic_uuid, NULL, GattCharacteristic::BLE_GATT_CHAR_PROPERTIES_NOTIFY  |
        GattCharacteristic::BLE_GATT_CHAR_PROPERTIES_WRITE |
        GattCharacteristic::BLE_GATT_CHAR_PROPERTIES_WRITE_WITHOUT_RESPONSE |
        GattCharacteristic::BLE_GATT_CHAR_PROPERTIES_READ ,NULL,0);

//GattCharacteristic *characteristics[] = {&writeChar,&readChar};
GattCharacteristic *characteristics[] = {&readChar};
GattService        customService(service_uuid, characteristics, sizeof(characteristics) / sizeof(GattCharacteristic *));

void disconnectionCallback(const Gap::DisconnectionCallbackParams_t *)
{
    BLE::Instance(BLE::DEFAULT_INSTANCE).gap().startAdvertising();
    led2 = 1;

}

void periodicCallback(void)
{
    led = !led; 

}

void writeCharCallback(const GattWriteCallbackParams *params)
{
    if(params->handle == readChar.getValueHandle()) {
      
        printf("Data received start: length = %d ",params->len);
        printf("\n\r");
        
         BLE::Instance(BLE::DEFAULT_INSTANCE).updateCharacteristicValue(readChar.getValueHandle(), readValue, DATA_MTU);

    }

}

void whenConnected(const Gap::ConnectionCallbackParams_t *)
{
    printf("whenConnected callack");
    printf("\n\r");
    led2 = 0;
}

void onDataSent(unsigned count)
{
    printf("onDataSent callack");
    printf("\n\r");
}

void gattServer_onUpdatesEnabled(GattAttribute::Handle_t handle)
{
    printf("gattServer_onUpdatesEnabled callack");
    printf("\n\r");
}

void gattServer_onUpdatesDisabled(GattAttribute::Handle_t handle)
{
    printf("gattServer_onUpdatesDisabled callack");
    printf("\n\r");
}

void gattServer_onConfirmationReceived(GattAttribute::Handle_t handle)
{
    printf("gattServer_onConfirmationReceived callack");
    printf("\n\r");
}

void bleInitComplete(BLE::InitializationCompleteCallbackContext *params)
{
    BLE &ble          = params->ble;
    ble_error_t error = params->error;

    if (error != BLE_ERROR_NONE) {
        return;
    }

    for (int i=0; i<16; i++) {
        service_data[i] = service_uuid.getBaseUUID()[i];
    }

    ble.gap().accumulateAdvertisingPayload(GapAdvertisingData::COMPLETE_LOCAL_NAME, (uint8_t *)DEVICE_NAME, sizeof(DEVICE_NAME));
    ble.gap().accumulateAdvertisingPayload(GapAdvertisingData::COMPLETE_LIST_128BIT_SERVICE_IDS, (uint8_t *)service_data, sizeof(service_data));
    ble.gap().setAdvertisingType(GapAdvertisingParams::ADV_CONNECTABLE_UNDIRECTED);
    ble.gap().setAdvertisingInterval(35); // multiples of 0.625ms ~ 22ms
    ble.gap().setTxPower(-18);

    ble.getPreferredConnectionParams(&fast);
    fast.minConnectionInterval = 16; // 20 ms
    fast.maxConnectionInterval = 32; // 40 ms
    fast.slaveLatency = 0;
    fast.connectionSupervisionTimeout = 10000;
    ble.addService(customService);

    ble.gap().onDisconnection(disconnectionCallback);
    ble.gap().onConnection(whenConnected);
    ble.gattServer().onDataWritten(writeCharCallback);
    ble.gattServer().onDataSent(onDataSent);

    ble.gattServer().onUpdatesEnabled(gattServer_onUpdatesEnabled);
    ble.gattServer().onUpdatesDisabled(gattServer_onUpdatesDisabled);
    ble.gattServer().onUpdatesDisabled(gattServer_onConfirmationReceived);

    ble.gap().startAdvertising();
}

int main(void)
{
    printf("\n\r********* Starting Main Loop *********\n\r");
    ticker.attach(periodicCallback, 1); /* Blink LED every second */

    BLE& ble = BLE::Instance(BLE::DEFAULT_INSTANCE);
    ble.init(bleInitComplete);

    while (ble.hasInitialized()  == false) {}

    while (true) {
        ble.waitForEvent();
    }
}

```

@ARMmbed/team-maximintegrated Please review this

closed 2018-11-16 by ciarmcom