ARMmbed/mbed-os #5492: Nordic bootloader support needed

dlfryar, 2017-11-13:
### Description
- Type: Enhancement
- Priority: Major

---------------------------------------------------------------

**Target**
nrf52 series (nrf52832 and nrf52840)

**Toolchain:**
GCC_ARM, ARM, IAR
----------------------------------------------------------------
## Enhancement

Nordic bootloader support needed

bootloader_util_reset()

https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_SDK11/libraries/bootloader_dfu/bootloader_util.c#L60

https://os.mbed.com/docs/v5.6/tutorials/creating-and-using-a-bootloader.html

dlfryar, 2017-11-13:
cc: @screamerbg

0xc0170, 2017-11-14:
@ARMmbed/team-nordic please review this request

nvlsianpu, 2017-11-14:
We are aware of lacking the generic bootloader support for nRF5x. We are analysing what and how support this needs so I expect a plan soon.

0xc0170, 2018-07-10:
This was resolved. I checked the master branch, both devices support bootloader.

closed 2018-07-10 by 0xc0170