ARMmbed/mbed-os #5493: Nordic STDIO_UART_*X moved from PeripheralNames to PinNames

dlfryar, 2017-11-13:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Enhancement

**Target**
nrf52 series

#define STDIO_UART_TX     TX_PIN_NUMBER
#define STDIO_UART_RX     RX_PIN_NUMBER

Should be moved in to target specific PinNames.h

https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF52840/PeripheralNames.h#L48

dlfryar, 2017-11-13:
cc: @screamerbg

0xc0170, 2017-11-14:
There are 3 blocks like the above in the TARGET_NRF5 directory.  The proposal would be to eliminate this duplication, and move this to common file for any TARGET_NRF5 ? Please send a pull request with proposal

cc @ARMmbed/team-nordic

pan-, 2017-11-14:
@dlfryar Could you explain the motivation of this request and why it should be in PinName.h and not PeripheralName.h ? Historically ` STDIO_UART_*X` has been defined in PeripheralNames.h and it is still the case today: There is 121 PeripheralNames.h  files defining ` STDIO_UART_*X` and only 8 PinName.h files defining it. 

More generally, the content that should be included and exposed by `objects.h`, `PeripheralNames.h` and `PinName.h` is not clearly identified (in the Handbook). It may be interesting to define formal requirements before opening issues against most of the vendor ports.

dlfryar, 2017-12-22:
I'll take a look at this further to see if I can isolate the issue to explain it further - stay tuned.