ARMmbed/mbed-os #5495: tests-mbed_hal-flash tests fails on EFM32PG_STK3401

studavekar, 2017-11-14:
### Description

Flash test fails for EFM32PG_STK3401 we would need this to be fixed for addition of device in CI.

- Type: Bug 
- Priority: Major 

```
[1510641210.53][CONN][INF] found KV pair in stream: {{__testcase_finish;Flash - erase sector;1;0}}, queued...
[1510641210.54][CONN][RXD] >>> Running case #4: 'Flash - program page'...
[1510641210.54][CONN][RXD] :205::FAIL: Element 0 Expected 206 Was 255
[1510641210.54][CONN][INF] found KV pair in stream: {{__testcase_start;Flash - program page}}, queued...
[1510641210.56][CONN][RXD] >>> 'Flash - program page': 0 passed, 1 failed with reason 'Assertion Failed'
[1510641210.56][CONN][RXD]
```
---------------------------------------------------------------
## Bug

**Target**
EFM32PG_STK3401

**Toolchain:**

GCC_ARM

**mbed-os sha:**
master

**Steps to reproduce**

mbed test --run -m EFM32PG_STK3401 -t GCC_ARM -n tests-mbed_hal-flash -v -V

**Logs 

Please rename to html [EFM32PG_STK3401.txt](https://github.com/ARMmbed/mbed-os/files/1469785/EFM32PG_STK3401.txt)

0xc0170, 2017-11-14:
cc @ARMmbed/team-silabs

stevew817, 2017-11-25:
@studavekar I can't reproduce on mbed-os master...

```
mbedgt: checking for GCOV data...
mbedgt: mbed-host-test-runner: stopped and returned 'OK'
mbedgt: test on hardware with target id: 203502D4000F442CD48277C3
mbedgt: test suite 'mbed-os-tests-mbed_hal-flash' .................................................... OK in 22.66 sec
	test case: 'Flash - buffer alignment test' ................................................... OK in 0.25 sec
	test case: 'Flash - clock and cache test' .................................................... OK in 0.08 sec
	test case: 'Flash - erase sector' ............................................................ OK in 0.03 sec
	test case: 'Flash - init' .................................................................... OK in 0.08 sec
	test case: 'Flash - mapping alignment' ....................................................... OK in 0.00 sec
	test case: 'Flash - program page' ............................................................ OK in 0.13 sec
mbedgt: test case summary: 6 passes, 0 failures
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.3143896557
mbedgt: test suite report:
+-------------------------+-----------------+------------------------------+--------+--------------------+-------------+
| target                  | platform_name   | test suite                   | result | elapsed_time (sec) | copy_method |
+-------------------------+-----------------+------------------------------+--------+--------------------+-------------+
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | mbed-os-tests-mbed_hal-flash | OK     | 22.66              | default     |
+-------------------------+-----------------+------------------------------+--------+--------------------+-------------+
mbedgt: test suite results: 1 OK
mbedgt: test case report:
+-------------------------+-----------------+------------------------------+-------------------------------+--------+--------+--------+--------------------+
| target                  | platform_name   | test suite                   | test case                     | passed | failed | result | elapsed_time (sec) |
+-------------------------+-----------------+------------------------------+-------------------------------+--------+--------+--------+--------------------+
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | mbed-os-tests-mbed_hal-flash | Flash - buffer alignment test | 1      | 0      | OK     | 0.25               |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | mbed-os-tests-mbed_hal-flash | Flash - clock and cache test  | 1      | 0      | OK     | 0.08               |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | mbed-os-tests-mbed_hal-flash | Flash - erase sector          | 1      | 0      | OK     | 0.03               |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | mbed-os-tests-mbed_hal-flash | Flash - init                  | 1      | 0      | OK     | 0.08               |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | mbed-os-tests-mbed_hal-flash | Flash - mapping alignment     | 1      | 0      | OK     | 0.0                |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | mbed-os-tests-mbed_hal-flash | Flash - program page          | 1      | 0      | OK     | 0.13               |
+-------------------------+-----------------+------------------------------+-------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 6 OK
mbedgt: completed in 25.17 sec
```

stevew817, 2017-12-01:
@studavekar Can you please verify your original findings and provide me with more details?

0xc0170, 2017-12-07:
> @studavekar Can you please verify your original findings and provide me with more details?

 @kegilbert

kegilbert, 2017-12-07:
@0xc0170 
Sorry for the late reply, the only board I was able to get my hands on is having some firmware/mounting issues. Will update when I get it working.

studavekar, 2017-12-19:
@stevew817 it fails for me on master. I not sure if i am missing something.
```
+-----------------+----------------------+---------------------------------------------------+---------------+--------------------------------------------------+-----------------+
| platform_name   | platform_name_unique | mount_point                                       | serial_port   | target_id                                        | daplink_version |
+-----------------+----------------------+---------------------------------------------------+---------------+--------------------------------------------------+-----------------+
| EFM32PG_STK3401 | EFM32PG_STK3401[0]   | /mnt/pci-0000_00_14_0-usb-0_5_5_1_3-scsi-0_0_0_0  | /dev/ttyACM4  | 2035022D000122D7D4751138                         | 0.14.3          |

```

```

[1513710164.30][CONN][RXD]
[1513710164.30][CONN][INF] found KV pair in stream: {{__testcase_finish;Flash - erase sector;1;0}}, queued...
[1513710164.31][CONN][RXD] >>> Running case #4: 'Flash - program page'...
[1513710164.31][CONN][RXD] :205::FAIL: Element 0 Expected 206 Was 255
[1513710164.31][CONN][INF] found KV pair in stream: {{__testcase_start;Flash - program page}}, queued...

mbedgt: test suite results: 1 FAIL
mbedgt: test case report:
+-------------------------+-----------------+----------------------+-------------------------------+--------+--------+---------+--------                                                                                                                                                                ------------+
| target                  | platform_name   | test suite           | test case                     | passed | failed | result  | elapsed                                                                                                                                                                _time (sec) |
+-------------------------+-----------------+----------------------+-------------------------------+--------+--------+---------+--------                                                                                                                                                                ------------+
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_hal-flash | Flash - buffer alignment test | 0      | 0      | SKIPPED | 0.0                                                                                                                                                                                |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_hal-flash | Flash - clock and cache test  | 0      | 0      | SKIPPED | 0.0                                                                                                                                                                                |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_hal-flash | Flash - erase sector          | 1      | 0      | OK      | 0.01                                                                                                                                                                               |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_hal-flash | Flash - init                  | 1      | 0      | OK      | 0.08                                                                                                                                                                               |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_hal-flash | Flash - mapping alignment     | 1      | 0      | OK      | 0.0                                                                                                                                                                                |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_hal-flash | Flash - program page          | 0      | 1      | FAIL    | 0.02                                                                                                                                                                               |
+-------------------------+-----------------+----------------------+-------------------------------+--------+--------+---------+--------  
```

0xc0170, 2018-01-08:
Same firmware, same board revision?

@studavekar Can you share the failed binary here that @stevew817 could rerun on his setup?

studavekar, 2018-01-10:
This is a configuration, same failure logs as in https://github.com/ARMmbed/mbed-os/issues/5495#issuecomment-352856719

MBED.HTM
```
<!-- mbed website redirect -->
<!-- Version: 0.14.3. build 557 -->
<html>
<head>
<meta http-equiv="refresh" content="0; url=http://mbed.org/device/?code=2035022D000122D7D4751138"/>
<title>mbed Website Shortcut</title>
</head>
<body></body>
</html>
```
git log -n1 --oneline
f10e663 Merge pull request #5807 from iiSeymour/patch-2

[image]

## Binary 
rename txt to bin
[flash.bin.txt](https://github.com/ARMmbed/mbed-os/files/1620431/flash.bin.txt)

studavekar, 2018-03-05:
closing this issue as its not seen release https://github.com/ARMmbed/mbed-os/releases/tag/mbed-os-5.7.6

closed 2018-03-05 by studavekar