ARMmbed/mbed-os #5496: rtos-mbed-timer and lp_ticker test faisl for TB_SENSE_12

studavekar, 2017-11-14:
### Description
tests-mbed_drivers-lp_ticker and tests-mbedmicro-rtos-mbed-timer tests fails for TB_SENSE_12 we would need this to be fixed for addition of device in CI.

- Type: Bug 
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
TB_SENSE_12 

**mbed-os sha:**
master 

**Steps to reproduce**

- mbed test --run -m TB_SENSE_12  -t GCC_ARM -n tests-mbedmicro-rtos-mbed-timer -v -V
- mbed test --run -m TB_SENSE_12  -t GCC_ARM -n tests-mbed_drivers-lp_ticker -v -V
**Logs 

Please rename to html 
[TB_SENSE_12.txt](https://github.com/ARMmbed/mbed-os/files/1469911/TB_SENSE_12.txt)

0xc0170, 2017-11-14:
cc @ARMmbed/team-silabs

stevew817, 2017-11-25:
@studavekar @0xc0170 This should be fixed by PR #5585

PR #5585 (2017-11-30): Fix issue with timer timebase on EFR32
## Description

The us_timer code was written based on MHz integer multiple HF clock frequencies, in order to optimize away some long division instructions. EFR32 doesn't conform to that (38.4), and so the timestamp ticks were off by 1%. Enough to trip up some CI tests on TB_SENSE_12 (resolves #5496)

## Status

**READY**
closed 2017-11-30 by 0xc0170