ARMmbed/mbed-os #5501: Ambiguous _tx_callback and _rx_callback complaints in eclipse for SerialBase.

darbyShaw, 2017-11-14:
### Description
- Type: Bug

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_F401RE

**Toolchain:**
GCC_ARM

**Toolchain version:**
6.3.1

**mbed-cli version:** 
1.2.0

**mbed-os version:**
5.6.4

**Expected behavior**
No errors should be present in SerialBase.cpp, there should be no ambiguity.

**Actual behavior**
Errors:
'_rx_callback' is ambiguous '
'_tx_callback' is ambiguous '
in SerialBase.cpp at:
```
SerialBase::SerialBase(PinName tx, PinName rx, int baud) :
#if DEVICE_SERIAL_ASYNCH
                                                 _thunk_irq(this), _tx_usage(DMA_USAGE_NEVER),
                                                 _rx_usage(DMA_USAGE_NEVER), **_tx_callback**(NULL),
                                                 **_rx_callback**(NULL),
#endif
                                                _serial(), _baud(baud) {
    // No lock needed in the constructor

    for (size_t i = 0; i < sizeof _irq / sizeof _irq[0]; i++) {
        _irq[i] = NULL;
    }

    serial_init(&_serial, tx, rx);
    serial_baud(&_serial, _baud);
    serial_irq_handler(&_serial, SerialBase::_irq_handler, (uint32_t)this);
}

```

**Steps to reproduce**
Export a project that uses serial interface from mbed-cli for gnuarmeclipse and compile.

----------------------------------------------------------------

## Question
The binary is compiled fine but eclipse still shows this error. Is this OK?

0xc0170, 2017-11-14:
What is your program, how to reproduce this? I do not remember seeing   this ambiguous error. I've just compiled blinky example for NUCLEO_F401RE , using GCC_ARM (latest master  b1e944f)

I assume your app is using Serial object or any other that inherits from SerialBase?

darbyShaw, 2017-11-15:
Yes you just initialize the Serial object and open the SerialBase.cpp and eclipse just marks an error there.

0xc0170, 2018-07-10:
Can you retry this with the latest release? If it is, please reopen.

closed 2018-07-10 by 0xc0170