ARMmbed/mbed-os #5502: Serial Communication problem in release profile build with Serial interface using aync_read.

darbyShaw, 2017-11-14:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
---------------------------------------------------------------
## Bug

**Target**
NUCLEO_F401RE

**Toolchain:**
GCC_ARM

**Toolchain version:**
6.3.1

**mbed-cli version:** 
1.2.0

**mbed-os version:**
5.6.4

**Expected behavior**
The async_read provided by SerialBase.cpp for Serial interface should read data always even with release builds.

**Actual behavior**
The async_read reads data once or twice initially and just stops reading data for release builds. But for Debug build it works completely fine all the time.

**Steps to reproduce**
Create a new project with following code in mbed-cli and export it for gnuarmeclipse. 
Compile and run and keep writing to the serial port periodically.
Try release and debug builds and compare.
Release build does not respond back with "got your data" more than half of the times but debug build responds all the time
```
#include "mbed.h"
Serial serialPort(SERIAL_TX, SERIAL_RX);
uint8_t buffer[500];
int data_available=1;

void rxCallback(int someEvent)
{
	data_available=1;
}
int main(void) {
    while(1)
    {
    	int ret = serialPort.read(&buffer[0], 200, rxCallback,
    			SERIAL_EVENT_RX_CHARACTER_MATCH|SERIAL_EVENT_RX_OVERFLOW|SERIAL_EVENT_RX_OVERRUN_ERROR, '}');
    	while(data_available==0);
    	data_available=0;
    	serialPort.puts("Got your data\r");
    }
}

```----------------------------------------------------------------

## Question
Why do the two builds behave differently?

0xc0170, 2017-11-14:
@ARMmbed/team-st-mcd

LMESTM, 2017-11-20:
@0xc0170  I don't think this is related to ST device.
@darbyShaw  the issue comes from your program. The compiler will optimize out some of the data_available usage if you're not telling the compiler that it can be changed from different contexts. 
Please use volatile declaration: 
```
volatile int data_available=1;
```
thanks to close the issue

closed 2017-11-20 by 0xc0170