ARMmbed/mbed-os #5504: C++11 format string being used

gjones1, 2017-11-15:
The following line us using %hh, this is a C++11 feature.  As I understand (reading the guidelines) the code should be GNU++98 (C++98 with extensions).  IAR7 is C++98 only and emits warnings.  Recommend changing or applying C++98 specific patch.

https://github.com/ARMmbed/mbed-os/blob/1394bf95f19fdb0cbf569359bbac0b02efec5d87/features/netsocket/SocketAddress.cpp#L70

0xc0170, 2017-11-15:
cc @geky @kjbracey-arm

kjbracey-arm, 2017-11-15:
This is a C99 feature, and we support C99 and are linking C99 and C++03 code into the same program, so this really has to work. There can't be two `scanf` definitions in the program - one for the C++03 and one for the C99. Any cases where C99 has an extension over C++03, we can assume it's present.

However I am aware that at least one of the libraries we use doesn't include C99 features in their printf/scanf - I've certainly seen diagnostics print literally '"%zu"' when they should be printing a size, and I've seen people rolling their own 64-bit print functions. (I think newlib-nano is a culprit?). So I wouldn't be surprised if this did fail.

Personally I'd rather fix the offending libraries and suppress this warning. But maybe we do have to avoid C99 prints for now. Not quite sure what this code does with newlib-nano.

pan-, 2017-11-17:
@kjbracey-arm newlib nano only supports conversion specifiers defined in the C89 standard ([1](https://github.com/32bitmicro/newlib-nano-1.0/blob/master/newlib/README.nano), [2](https://answers.launchpad.net/gcc-arm-embedded/+question/257014)). 

Alignement between C and C++ compilers can be better if we compile C++ code in C++11 mode; the upcomming compiler changes may be the good oportunity to improve consistentency (please see #5507).

kjbracey-arm, 2018-06-20:
This was resolved by https://github.com/ARMmbed/mbed-os/pull/6524

closed 2018-06-20 by kjbracey-arm