ARMmbed/mbed-os #5506: MBRBlockDevice::partition - type

JaniSuonpera, 2017-11-15:
### Description
- Type: Question

## Question
I am looking documentation https://os.mbed.com/docs/v5.6/mbed-os-api-doxy/class_m_b_r_block_device.html#ace6dd73e8514a0a2afab8f8bc37bada8

I tried to find supported values/filesystems for type, which is "8-bit partition type to identitfy partition's contents". Would you clarity usage of parameter type with MBRBlockDevice::partition? What are supported values? Is it type same than Partition ID in table https://en.wikipedia.org/wiki/Partition_type#List_of_partition_IDs?

0xc0170, 2017-11-15:
cc @geky @deepikabhavnani

JanneKiiskila, 2017-11-27:
There really should be some kind of enum or similar to specify this parameter.

geky, 2018-02-15:
Sorry for such a big delay in getting to these issues.

Yes, the type parameter is an ID from that list:
https://en.wikipedia.org/wiki/Partition_type#List_of_partition_IDs

I didn't include it initially because it's such an unwieldy list (>200 options), but if it helps I can add an enum of the most common types.

I'm also thinking it would be a good idea to just use 0x83 as a default.

bulislaw, 2018-08-13:
@JaniSuonpera @geky can we close it?

JaniSuonpera, 2018-08-14:
Yes, this can be closed. I propose that we clarify mbed-os documentation about usage of partition type.

dannybenor, 2018-08-21:
@JaniSuonpera can you please close this?

deepikabhavnani, 2018-08-21:
I have created docs issue, hence should be good to close https://github.com/ARMmbed/mbed-os-5-docs/issues/673

closed 2018-08-21 by deepikabhavnani

JaniSuonpera, 2018-08-22:
Yes, this can be closed. I agree that ARMmbed/mbed-os-5-docs#673 replace this one.