ARMmbed/mbed-os #5517: lib_wlan_mbed_gcc.a is 5x bigger than lib_wlan_mbed_arm.ar

xiangzhai, 2017-11-17:
Hi ARMmbed developers,

### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
REALTEK_RTL8195AM

**Toolchain:**
GCC_ARM|ARM

**Toolchain version:**
gcc version 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437] (GNU Tools for ARM Embedded Processors 6-2017-q2-update)

ARM Compiler v6.9 based on LLVM 6

**mbed-cli version:** 
(`mbed --version`)
1.2.2

**mbed-os sha:**
(`git log -n1 --oneline`)
6720bfd

**Expected behavior**
Build lib_wlan_mbed_gcc.a turn off `-g` and open `-O3` optimazation for GNU Toolchain, or provide the wlan library source code just like [ST DISCO_L475VG_IOT01A](https://os.mbed.com/teams/ST/code/DISCO_L475VG_IOT01A_wifi/).

**Actual behavior**
Bring in `.debug_str`, `.debug_abbrev`, `.debug_info`, `.debug_frame`, `.debug_line` sections, so the size is **5x** bigger than lib_wlan_mbed_arm.ar built with ARM Compiler.

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**
[Size and performance](http://www.leetcode.cn/2017/11/build-mbed-with-llvm-toolchain.html) is crital for Embedded usecase. 

```
   text    data     bss     dec     hex filename
 276328   50000  473484  799812   c3444 mbed-os-example-wifi.elf
```

compare with

```
   text    data     bss     dec     hex filename
  71584    2688   11880   86152   15088 DISCO_L475VG_IOT01-wifi_http_server.elf
```

**Suggested enhancement**
Provide the source code just like ST DISCO_L475VG_IOT01A.

Regards,
Leslie Zhai - a LLVM developer https://reviews.llvm.org/p/xiangzhai/

0xc0170, 2017-11-17:
@ARMmbed/team-realtek

pan-, 2017-11-17:
@xiangzhai Not sure to understand what the problem is;  `.debug_str`, `.debug_abbrev`, `.debug_info`, `.debug_frame` and `.debug_line` sections may be in the elf but are not part of the binary flashed on boards.

Your also comparing apple and oranges: it is innane to compare `DISCO_L475VG_IOT01-wifi_http_server` with `mbed-os-example-wifi` on the REALTEK_RTL8195AM; those are different examples and different examples. 

I benchmarked mbed-os-example-wifi on REALTEK_RTL8195AM in release here are the results:

**gcc** 

```
Total Static RAM memory (data + bss): 92468 bytes
Total Flash memory (text + data): 284148 bytes
```

**ARM compiler 5.06**

```
Total Static RAM memory (data + bss): 123345 bytes
Total Flash memory (text + data): 325754 bytes
```

As you can see, GCC produce a more compact and less memory hungry binary. 

Last but not least, I would be interrested to understand how you compiled `mbed-os-example-wifi.elf` with ARM compiler 6 ? It requires an update of the realtek library which has not been delivered yet ...

xiangzhai, 2017-11-17:
@pan- The debug section not in ELF or BIN, and yes, DISCO_L475VG_IOT01 https://os.mbed.com/teams/ST/code/DISCO_L475VG_IOT01A_wifi/  is **10x** smaller than Realtek, it is able to extract the static library, then objdump to see. I am using patched LLVM Toolchain to build WiFi example http://www.leetcode.cn/2017/11/build-mbed-with-llvm-toolchain.html Workaround patch https://github.com/xiangzhai/ufiasco

xiangzhai, 2017-11-17:
https://github.com/xiangzhai/ufiasco/commit/acd4e9c9f5f2fb71a24bd2d5892e8447b52827af

and

https://github.com/xiangzhai/ufiasco/commit/da062a1af034fa9711413a5038a87dc17b645e13

Based on ARM C MicroLib, but I am trying to use others https://keithp.com/blogs/embedded-arm-libc/

pan-, 2017-11-17:
@xiangzhai We do have a branch for ARM compiler 6 support, at the current minute armlink rejects the precompiled library from realtek because it has been compiled for ARM compiler 5; I would be happy to compare GCC and ARM compiler numbers once the realtek library is available. I'm pretty sure difference between the two won't be in the 5X ballpark giben that GCC outperform ARM compiler 5.06 on this one and usually Arm compiler 5.06 outperform Arm compiler 6.X.

Code comparison between the DISCO_L475VG_IOT01A_wifi and the REALTEK module is a complete different issue. You can open a new issue if you judge that size of the realtek library is not sustainable but this does not relate to GCC or ARMCC.

xiangzhai, 2017-11-17:
@pan- salute for your hard work!

PS: it is glad to see more Realtek boards will support ARMmbed :)

xiangzhai, 2017-11-23:
ellcc Toolchain is awesome! http://ellcc.org/viewvc/svn/ellcc/trunk/libecc/patches/

closed 2017-11-23 by xiangzhai