ARMmbed/mbed-os #5522: thread max_stack() returns always aligned stacksize-4 in rtos5

karsev, 2017-11-17:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
ARM

**Toolchain version:**
5.06
**mbed-cli version:** 
1.1.1

**mbed-os sha:**
78474a512 

**Expected behavior**
Should return maximum stack memory usage to date

**Actual behavior**
Calling max_stack() returns: aligned stacksize-4
**Steps to reproduce**

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

**Suggested enhancement**
thread stack should be filled with the osRtxStackFillPattern pattern instead of hardcoded 0xE25A2EA5U.
rtx_thread.c already sets the first byte to osRtxStackMagicWord (0xE25A2EA5U) and the rest is filled with the osRtxStackFillPattern.

0xc0170, 2017-11-21:
This is the code for max stack in Thread.cpp (https://github.com/ARMmbed/mbed-os/blob/master/rtos/Thread.cpp#L292). 

I do not recall why are are not using by default the `osThreadGetStackSize` and `osThreadGetStackSpace`. I assume as we fill the stack with our own defined value (Thread::start method does it) therefore this is used instead. If these functions are used instead, does it return the expected value?

If stack stats are enabled, then `OS_STACK_WATERMARK` is set to 1 , and RTX watermark is used.

cc @c1728p9

c1728p9, 2017-11-21:
The guard word might be causing max_stack to be returning the wrong value. 

@0xc0170 if osThreadGetStackSize and osThreadGetStackSpace are used it should fix the problem, as long as watermarking is always on. Here is the previous reason why this was left out:
https://github.com/ARMmbed/mbed-os/pull/4776#issuecomment-324052011

kegilbert, 2018-07-02:
@c1728p9 Picked this up from the TRIAGE bot, are the comments you linked to still valid/what should be done with this issue?