ARMmbed/mbed-os #5524: tests: Unable to share code in tests

geky, 2017-11-17:
I realize this issue probably already exists, but I can't seemt to find it.

Currently, we have no method for sharing code between related tests. Currently the options are:
1. Copy test code into every test directory
2. Include the code at the top level and compile always

Option 1 violates the principle of don't repeat yourself. Leads to a lot of code duplication that can fall out of sync as tests are modified.

Option 2 leads to several minor issues:
- Where do we store the common code? Currently common tests code is being stored inconsistenly. We've started using TEST_COMMON, but this could lead to compatibility concerns if the name becomes used in the mbed tools.
- Additional compile time, mbed doesn't need help with this
- Distributed dependencies makes debugging tougher when jumping between files
- Test code needs additional ifdefs to avoid breaking the codebase if there is a compile issue outside of testing.

We already have the special folder host_tests in the testing folder layout. It would be helpful to have something similar for shared code. (maybe TESTS/filesystem/UTIL?).

cc @studavekar, @theotherjimmy, @c1728p9, @bridadan

theotherjimmy, 2017-11-17:
Definitely a problem. I don't presently have a design for a solution. I'm very open to suggestions here.

bridadan, 2017-11-17:
Yep, this was a bit of an oversight when this was first implemented!

> We already have the special folder host_tests in the testing folder layout. It would be helpful to have something similar for shared code. (maybe TESTS/filesystem/UTIL?).

Seems ok, and should be pretty straight forward to accomplish. The edits would need to be made in `tools/test.py` and `tools/test_api.py`. I'd definitely make it a capital directory to fit in with our existing "All caps directories are special". Just needs to be documented in Mbed CLI's readme.

0xc0170, 2017-11-21:
A set of tests (test unit) might need a common folder, as shown above `TESTS/filesystem/UTIL`. How these test units can specify the common folder dependency? For filesystem it can be util, for spi drivers another one, tickers another one. 

How can this be defined?  Set somehow in our tools as @bridadan proposed above? Who should define this dependency? A target defines it's own dependencies via labels. Can one test unit do the same? Not a config option?

bridadan, 2017-11-21:
> Can one test unit do the same

Not to my knowledge, no. And there isn't really anything in the tools right now that would allow us to do that.

> Not a config option

Same thing here, there isn't really a mechanism for this that I know of.

Since we already treat the folders `TESTS/host_tests` and `TESTS/<test group>host_tests` differently, would anyone be opposed to adding a similar folder? I'd propose `TESTS/common` and `TESTS/<test group>/common`.

geky, 2018-01-09:
Here's the conception of @bridadan's thoughts:
https://github.com/ARMmbed/mbed-os/pull/5819

closed 2018-01-18 by cmonr