ARMmbed/mbed-os #5529: tests-mbed_drivers-ticker  tests are failing on KL46Z

studavekar, 2017-11-19:
### Description

tests-mbed_drivers-ticker tests are failing on KL46Z 

```
[1511076881.05][CONN][RXD] >>> Running case #9: 'Test multi ticker'...
[1511076881.09][CONN][INF] found KV pair in stream: {{__testcase_start;Test multi ticker}}, queued...
[1511076881.22][CONN][RXD] :195::FAIL: Expected 16 Was 6
[1511076881.28][CONN][INF] found KV pair in stream: {{__testcase_finish;Test multi ticker;0;1}}, queued...
[1511076881.35][CONN][RXD] >>> 'Test multi ticker': 0 passed, 1 failed with reason 'Assertion Failed'
```

- Type: Bug 

**Target**
KL46Z

**Toolchain:**
GCC_ARM|ARM|IAR

**Steps to reproduce**
mbed test -m KL46Z -t GCC_ARM-n tests-mbed_drivers-ticker -vv

0xc0170, 2017-11-20:
@maciejbocianski Can you please look at t his one. Is this related to https://github.com/ARMmbed/mbed-os/pull/5261 ?

cc @ARMmbed/team-nxp

maciejbocianski, 2017-11-21:
@0xc0170  This issue was already reported see #KL46Z

0xc0170, 2017-11-21:
OK,I can see now, this was originally reported here https://github.com/ARMmbed/mbed-os/issues/5307.

@maciejbocianski if we can rework the original report to be clear it has KL46Z in there, this can be closed as duplicate.

0xc0170, 2018-07-10:
#5261 resolved it

closed 2018-07-10 by 0xc0170