ARMmbed/mbed-os #5537: Tools:  multiple region - tools error msg improvement

0xc0170, 2017-11-20:
### Description
- Type: Enhancement 
- Related issue: `#5021`
- Priority:  Minor

Referenced PR above , there is this request:

The problem

> ELF files containing multiple regions were being converted to a folder with each region as a separate file, this caused the build scripts to fail since the .bin was not created.

Possibly enhancement

> Maybe the mbed-os build tools should print a nicely formatted error message if this occurs and there isn't a post build hook for the platform

@theotherjimmy Please add more information if needed

theotherjimmy, 2017-11-20:
@0xc0170 That looks correct. I changed the related issue field so that it links correctly.

cmonr, 2018-07-12:
Closing this since referred PR has since been closed.

closed 2018-07-12 by cmonr