ARMmbed/mbed-os #5542: Ethernet driver in mbed/driver does not call ST specific HAL libs?

peeyushpashine, 2017-11-21:
### Description
- Type: Question

**Target**
TARGET_STM

Ethernet driver in mbed-os/driver does not call ST specific HAL libs while these HAL functions are implemented for other targets.

I see [here](https://github.com/ARMmbed/mbed-os/tree/c0fe3b510931e2e60fc40b6ccb52a39ba31c5e00/features/FEATURE_LWIP/lwip-interface) EthernetInterface.cpp implements target specific arch based LWIP interface for ST. But what is the reason for keeping driver file here insread of mbed-os/driver ?
@0xc0170 @bcostm @geky

0xc0170, 2017-11-21:
@peeyushpashine mbed-os/driver, do you mean `Ethernet` class?

peeyushpashine, 2017-11-22:
@0xc0170 Yes the Ethernet class.

0xc0170, 2017-11-22:
I believe this is duplicate of https://github.com/ARMmbed/mbed-os/issues/4586 that you reported earlier. Similar question. The answer is that ethernet in device_has is deprecated, thus also this class. However we have this subject again, and I noticed Ethernet class is still present in the drivers without any indication of deprecation.

@kjbracey-arm @mikaleppanen  @geky If the above is correct, we might deprecate the class, and later remove it ?

kjbracey-arm, 2017-11-22:
This is correct - the class should be deprecated.

Work in progress is refining the EMAC APIs, so that `EthernetInterface` is generalised from "Ethernet using TARGET_XXX's native lwIP driver" into "Ethernet using any EMAC driver attached to any on-board IP stack". Last draft of that was at #5202, next one coming soon. Main difference since 5202 is that the EMAC API is C++-ified so that an EMAC driver is an implementation of an abstract `class EMAC`.

In the new model anyone can implement `EMAC` objects, and a target (or other) can nominate an EMAC object as the default for `EthernetInterface`.

This effectively pulls all Ethernet support out of the HAL, putting it into the C++ domain. My plan is that a target would make its Ethernet available by implementing `EMAC &EMAC::get_default_instance()`.

We may retain `DEVICE_EMAC` to indicate an on-board default EMAC is provided, but lack of on-board target default doesn't preclude an application instantiating `EthernetInterface(MyEMACModule)`.

cmonr, 2018-07-12:
@kjbracey-arm Pretty sure this can be closed. Mind confirming?

0xc0170, 2018-07-25:
EMAC feature was integrated. This shall be closed

closed 2018-07-25 by 0xc0170