ARMmbed/mbed-os #5543: Test compilation issue for LWIP targets

jeromecoutant, 2017-11-21:
Hi

Tests from ci-test-shield can not be executed any more with mbed-os-5.6.5 for targets which are supporting LWIP.... :-(

ex:
mbed test -t GCC_ARM -m NUCLEO_F207ZG -n tests* --compile
...
        [Error] Mixed.cpp@68,22: 'MBED_CONF_APP_I2C_SDA' was not declared in this scope
        [Error] Mixed.cpp@68,44: 'MBED_CONF_APP_I2C_SCL' was not declared in this scope
        [Error] Mixed.cpp@68,66: 'MBED_CONF_APP_I2C_EEPROM_ADDR' was not declared in this scope
...

It seems that guilty commit is 4a5ad4ee3ebb34b173bf8753f0670c03ecc0f835

Thx

0xc0170, 2017-11-21:
@jeromecoutant I do not see the connection between i2c pins and the referenced PR that is changing some networking tests. Could you clarify how did you get to that guilty commit? It would be helpful

cc @yennster @sarahmarshy

jeromecoutant, 2017-11-21:
> Could you clarify how did you get to that guilty commit?

Before, compilation is OK. With the commit, it fails.... :-(
It looks like with the patch, mbed_app.json file from ci-test-shield is no more taken into account.
I could add some print in the py file to check...

jeromecoutant, 2017-11-21:
> I could add some print in the py file to check

Issue looks to start in test.py just after this:

        # Assign config file. Precedence: test_config>app_config
        # TODO: merge configs if both given

Maybe it's time to implement the todo :-)
When there is no LWIP, config is None.
With LWIP target, config is EthernetInterface.json file.

geky, 2017-11-21:
It looks like the test config is completely overriding the mbed_app.json, @sarahmarshy, @studavekar, is this expected?

sarahmarshy, 2017-11-21:
No. Definitely a bug. Will need to have some discussion with Shrikant about correct behavior for this.

yennster, 2017-11-22:
@jeromecoutant As a workaround, have you tried appending the location of your mbed_app.json file to your test command? 
i.e. `mbed test -t GCC_ARM -m NUCLEO_F207ZG -n tests* --app-config mbed_app.json --compile`

I am looking into a fix for test.py so it will not always override the mbed_app.json file with the test config.

jeromecoutant, 2017-11-23:
Hi
yes,
mbed test -t GCC_ARM -m NUCLEO_F207ZG -n tests* --app-config mbed_app.json --compile
is working!
Thx

Let's try now #5559

closed 2017-12-20 by 0xc0170

jeromecoutant, 2017-12-20:
Closed but #5559 will be available only in release-version: 5.8.0-rc1 ...

I will apply the proposed workaround.

PR #5559 (2017-12-20): Fix for default test config file
## Description

Fixes #5543

This change allows for `mbed test` to build with an `mbed_app.json` file by default if one exists in the project's root directory (i.e. for running the [ci-test-shield tests](https://github.com/ARMmbed/ci-test-shield)). However, if the `mbed_app.json` file is not in the root directory, then it's location needs to be specified in the command with the `--app-config` parameter.

## Steps to test or reproduce

**Before fix:**
If you run the [ci-test-shield tests](https://github.com/ARMmbed/ci-test-shield) with `mbed test -t GCC_ARM -m N [... cut]