ARMmbed/mbed-os #5544: Add HAL interface for retrieving Unique Serial Number of CPU

MikeDK, 2017-11-21:
### Description
- Type: Enhancement

## Enhancement

**Reason to enhance**
As far as I know there is no abstract layer to retrieve a CPU unique serial number which is implemented by nearly all manufacturers.
At the moment a developer has to write native board/cpu code for getting the unique serial number.

**Suggested enhancement**
Add new HAL interface for getting unique serial number, which shall then be implemented by the different board/cpu manufacturers or some other willing persons ;)

Also add a new class in drivers, which will use the hal interface.
Can be some very simple class which reads unique serial number via hal, stores it in a byte array of variable length (some manufacturers have 8 byte uid, some have 12 bytes, ...) and some type cast operators which return the uid in different formats (byte array, std::vector<byte>, std::string, const char*, ...)

I have already implemented such a class, which works with EFM32, STM32, and NRF5x boards/cpus.
But this class directly calls code provided by the different manufacturer's SDKs.
It would be more reasonable to implement a real HAL interface for this.

**Pros**
For writing portable mbed code which needs some kind of unique device id, this is a must-have.

0xc0170, 2017-11-21:
> I have already implemented such a class, which works with EFM32, STM32, and NRF5x boards/cpus.

Can you reference the code ? Or paste here code snippet ?

cc @bulislaw @c1728p9 @scartmell-arm

MikeDK, 2017-11-21:
ok no problem, I can paste it...:

UniqueSerial.h
`#include <mbed.h>
#include <string>

class UniqueSerial
{
public:
    UniqueSerial();
    virtual ~UniqueSerial();
    
    operator std::string() { return uidString; }
    operator const char*() { return uidString.c_str(); }
    
    
private:
    std::string uidString;
    
};
`

UniqueSerial.cpp
`#include <cstdio>
#include "UniqueSerial.h"

#if defined(TARGET_EFM32)
#include "em_system.h"
#elif defined(TARGET_STM32F0)
#include "stm32f0xx_ll_utils.h"
#elif defined(TARGET_STM32F1)
#include "stm32f1xx_ll_utils.h"
#elif defined(TARGET_STM32F2)
#include "stm32f2xx_ll_utils.h"
#elif defined(TARGET_STM32F3)
#include "stm32f3xx_ll_utils.h"
#elif defined(TARGET_STM32F4)
#include "stm32f4xx_ll_utils.h"
#elif defined(TARGET_STM32F7)
#include "stm32f7xx_ll_utils.h"
#elif defined(TARGET_STM32L0)
#include "stm32l0xx_ll_utils.h"
#elif defined(TARGET_STM32L1)
#include "stm32l1xx_ll_utils.h"
#elif defined(TARGET_STM32L4)
#include "stm32l4xx_ll_utils.h"
#endif

UniqueSerial::UniqueSerial()
{
#if defined(TARGET_EFM32)
    char buf[17];
    uint64_t uid = SYSTEM_GetUnique();
    snprintf(buf, 17, "%.16llX", uid);
#elif defined(TARGET_FAMILY_STM32)
    char buf[25];
    uint32_t uid0 = LL_GetUID_Word0();
    uint32_t uid1 = LL_GetUID_Word1();
    uint32_t uid2 = LL_GetUID_Word2();
    snprintf(buf, 25, "%.8lX%.8lX%.8lX", uid2, uid1, uid0);
#elif defined(TARGET_NRF5)
    char buf[17];
    snprintf(buf, 17, "%.8lX%.8lX", NRF_FICR->DEVICEID[1], NRF_FICR->DEVICEID[0]);
#endif
    uidString = buf;
}

UniqueSerial::~UniqueSerial()
{

}
`

This is just a very basic class ... I also should not store the uid in a std::string in ascii format, as this needs twice the ram ;)

Basically this works, but I think to be more mbed-like, there should be a HAL interface for it, so the UniqueSerial class won't have to call native code directly.

MikeDK, 2017-11-21:
ok the "insert code" button here doesn't work for me ;)

i'll attach the files instead...

[UniqueSerial.zip](https://github.com/ARMmbed/mbed-os/files/1491190/UniqueSerial.zip)

bulislaw, 2017-11-21:
We have it in our plans, but not yet on the roadmap. If you need it anytime soon I would recommend to try to contribute this functionality to mbed so the whole community can benefit.

MikeDK, 2017-11-22:
Ok I implemented the functionality and also made the implementation for STM32, EFM32 and NRF5x targets.

Pull Request: #5557

Someone then has to urge all the other vendors to also implement this ;-)

cmonr, 2018-07-12:
Closing since this appears to have been resolved with the merged PR.

closed 2018-07-12 by cmonr

Jonathanchan1996, 2024-09-19:
> the "insert code" button here doesn't work for me ;)
> 
> i'll attach the files instead...
> 
> [UniqueSerial.zip](https://github.com/ARMmbed/mbed-os/files/1491190/UniqueSerial.zip)

man, we are already on Github. How come u zip your code....