ARMmbed/mbed-os #5545: Pin definition A3 missing on REALTEK_RTL8195

SeppoTakalo, 2017-11-21:
### Description
- Type: Bug
- Related issue: `#abc`
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
REALTEK_RTL8195A

**Toolchain:**
GCC_ARM

**Expected behavior**
Arduino pin names should be defined.

**Actual behavior**
Pin A3 is missing from:
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Realtek/TARGET_AMEBA/TARGET_RTL8195A/PinNames.h#L175

0xc0170, 2017-11-21:
@ARMmbed/team-realtek

JanneKiiskila, 2017-11-28:
This will be blocking the support for mbed-os-example-client and Cloud Client for Realtek board.

JanneKiiskila, 2017-11-28:
https://github.com/ARMmbed/mbed-os/pull/5611 PR created to get rid of this problem (at least temporarily). Realtek should review and comment if the NC is a suitable solution for A3. At least it enables compilation again with ST-WIFI-X module included (see tip of master of `mbed-os-example-client`).

Archcady, 2017-11-29:
There is no A3~A5 pins in our Realtek AMEBA board and hence these pins are mapped to NC in mbed_app.json. This is a hardware limitation of our platform. I think this is the best solution from our side.

JanneKiiskila, 2017-11-29:
The mbed_app.json doesn't help with the comilation issues, those needs the be in the pinnames.h.

SeppoTakalo, 2017-11-29:
Proper fix would be https://github.com/ARMmbed/mbed-os/pull/5611 because if the board claims to be `ARDUINO` form factor then all the Arduino pin names should be defined. Otherwise you would break build of some drivers relying on the names.

Archcady, 2017-11-30:
I have approved the fix #5611. I think it can be merged now

JanneKiiskila, 2017-12-03:
@SeppoTakalo  - I think the issue can be closed now?

closed 2017-12-04 by 0xc0170