ARMmbed/mbed-os #5551: retarget: ftell seems to be calling file_seek instead of file_tell

geky, 2017-11-22:
This is not incorrect behaviour, but is a hit to performance.

The seek function can be used to get the needed info for ftell, but a call to seek has a lot more implications since it may change the file position. Most notable caches may end up flushed to disk to prepare for the jump.

We have file_tell in the filesystem API, so ftell should be calling that instead. In theory, we just need to look into the standard library to figure our where the correct hook is.

cc @dannybenor, @davidsaada

dannybenor, 2017-11-22:
https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/FileSystem.cpp#L64
off_t FileSystem::file_tell(fs_file_t file)
{
return file_seek(file, 0, SEEK_CUR);
}

geky, 2017-11-22:
@dannybenor, file_seek is declared virtual [[1]](https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/FileSystem.h#L183).

The specific filesystem implementation is free to override this:
- [FATFileSystem::file_tell](https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/fat/FATFileSystem.cpp#L575-L583) 
- [LittleFileSystem::file_tell](https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/fat/FATFileSystem.cpp#L575-L583)

theamirocohen, 2018-08-30:
After looking in to this issue I found that the behavior mentioned is hidden in the compilers implementation.
IAR and GCC_ARM retarget ftell through lseek method in mbed_retarget.cpp, and ARM doesn't even do that (probably handles it internally).
It seems like the changes required should involve the compilers code, and only after these changes we can consider implementing the ftell method in mbed_retarget.

geky, 2018-08-30:
Actually ARMCC probably has the best implementation. It's perfectly fine for ftell to just report a cached file position, and by not calling the retargeted seek function, ftell would be faster and avoid the flushing logic.

closed 2018-12-04 by ciarmcom