ARMmbed/mbed-os #5564: KL46Z: Inconsistent Ticker callback calls when scheduled many tickers at the same time

maciejbocianski, 2017-11-23:
When schedule many Tickers one after the other with the same time interval 100ms and then wait 105ms it's expected that after this time all Tickers will be fired once. But we get other results sometimes not all tickers are fired and sometimes some fire more then once

Problem noticed in https://github.com/ARMmbed/mbed-os/pull/5261 in the following test
```
/** Test many tickers run one after the other

    Given many Tickers
    When schedule them one after the other with the same time intervals
    Then tickers properly execute callbacks
    When schedule them one after the other with the different time intervals
    Then tickers properly execute callbacks
 */
void test_multi_ticker(void)

```
Sample code for reproduction
```
#define MULTI_TICKER_TIME_MS 100
#define TICKER_COUNT 16
Ticker ticker[TICKER_COUNT];
uint32_t multi_counter;

void increment_multi_counter(void)
{
   core_util_atomic_incr_u32(&multi_counter, 1);
}

const uint32_t extra_wait = 5; // extra 5ms wait time
multi_counter = 0;
for (int i = 0; i < TICKER_COUNT; i++) {
    ticker[i].attach_us(callback(increment_multi_counter), MULTI_TICKER_TIME_MS * 1000);
}

Thread::wait(MULTI_TICKER_TIME_MS + extra_wait);
for (int i = 0; i < TICKER_COUNT; i++) {
    ticker[i].detach();
 }

TEST_ASSERT_EQUAL(TICKER_COUNT, multi_counter);
```

### Description
- Type: Bug
- Related PR: https://github.com/ARMmbed/mbed-os/pull/5261
- Related Issues: https://github.com/ARMmbed/mbed-os/issues/5307 https://github.com/ARMmbed/mbed-os/issues/5309 https://github.com/ARMmbed/mbed-os/issues/5529
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
KL46Z

**Toolchain:**
GCC_ARM|ARM|IAR

**mbed-os sha:**
master

**Expected behavior**
Ticker callback is executed exactly 16 times

**Actual behavior**
Ticker callback is executed several times 3, 7, 10, 32  but not as expected

0xc0170, 2017-11-24:
@ARMmbed/team-nxp please review

0xc0170, 2017-11-24:
Is this similar to https://github.com/ARMmbed/mbed-os/issues/5529 ?

maciejbocianski, 2017-11-28:
Yes it's the same problem

0xc0170, 2017-11-28:
OK, closing this as duplicate. Please refer to 5529.

closed 2017-11-28 by 0xc0170